espressif/esp-who #97: matrix3du item alloc failed when working with AmazonFreeRTOS

Ascarron, 2019-07-31:


XiaochaoGONG, 2019-08-05:
Not sure if the os support mapping external memory to heap. You can try to allocate about 300KB space to see if SPI RAM works.

Ascarron, 2019-08-05:
Yes the mapping to the external memory works. To test this I did
`image_matrix = dl_matrix3du_alloc(1, 160, 120, 3);`
and the memory is allocated with a decrease in free heap size.

XiaochaoGONG, 2019-09-04:
160x120x3 = about 57KB, which is still allocated in the internal memory. It seems you didn't successfully enable the SPI RAM. You may transfer this issue to [esp-idf](https://github.com/espressif/esp-idf).

fash-aryan, 2021-08-31:
Has there been any updates on this issue and how to solve it? I am having the same problem when allocating 57KB of memory.