espressif/esp-who #106: Bug in dl_lib_matrix3d

geekboood, 2019-08-09:
Hi, I encounter a bug when I try to use your DL lib. I set a conv kernel of shape (16, 3, 3, 3), which means 16 filters each with 3x3 kernel and 3 channels and an image of shape (1, 14, 14, 3) and the result is different from PyTorch. Code is here. 
`#include <Arduino.h>
#include <dl_lib.h>
#include <weight.h>

dl_matrix3d_t *ConvBNReLU(dl_matrix3d_t *in, dl_matrix3d_t *filter, dl_matrix3d_t *bnw, dl_matrix3d_t *bnb, int inp, int outp, int kernel, int stride, int padding, int group) {
  dl_matrix3d_t *bias = dl_matrix3d_alloc(1, 1, 1, outp);
  dl_matrix3d_t *result;
  if (group != inp) {
    result = dl_matrix3d_conv(in, filter, bias, stride, stride, padding, 1);
  }
  else {
    result = dl_matrix3d_depthwise_conv(in, filter, stride, stride, padding, 1);
  }
  dl_matrix3d_batch_normalize(result, bnw, bnb);
  dl_matrix3d_relu(result, 6);
  return result;
} 

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println(ESP.getFreeHeap());
  dl_matrix3d_t *kernel, *img;
  kernel = dl_matrix3d_alloc(16, 3, 3, 3);
  img = dl_matrix3d_alloc(1, 14, 14, 3);
  // free(kernel->item);
  // free(img->item);
  kernel->item = (fptp_t *)&weight;
  img->item = (fptp_t *)&image;
  dl_matrix3d_t *bias = dl_matrix3d_alloc(1, 1, 1, 16);
  dl_matrix3d_t *result = dl_matrix3d_conv(img, kernel, bias, 2, 2, 1, 1);
  Serial.printf("%d %d %d %d\n", result->n, result->h, result->w, result->c);
  Serial.printf("%f %f %f %f\n", result->item[0], result->item[1], result->item[2], result->item[3]);
  dl_matrix3d_print(result, "");
  
}

void loop() {
  // put your main code here, to run repeatedly:
}`
kernel matrix
`fptp_t weight[16][3][3][3] = {1., 8., 1., 0., 5., 8., 9., 5., 9., 5., 4., 7., 9., 8., 8., 5., 1.,
       5., 3., 9., 3., 0., 5., 3., 2., 4., 6., 0., 6., 8., 3., 5., 3., 2.,
       9., 6., 4., 7., 5., 5., 8., 2., 2., 1., 4., 5., 6., 5., 7., 5., 4.,
       1., 9., 1., 1., 0., 5., 5., 1., 4., 9., 5., 0., 0., 5., 2., 7., 6.,
       1., 0., 1., 1., 7., 3., 9., 6., 5., 3., 4., 4., 5., 0., 3., 5., 0.,
       5., 8., 2., 4., 6., 5., 7., 1., 1., 6., 5., 9., 2., 8., 1., 4., 1.,
       0., 5., 6., 0., 3., 8., 0., 9., 2., 8., 7., 5., 8., 0., 3., 4., 5.,
       9., 7., 9., 5., 8., 8., 2., 2., 1., 6., 0., 6., 5., 8., 4., 4., 1.,
       4., 7., 4., 6., 9., 0., 0., 0., 2., 7., 5., 3., 1., 7., 3., 9., 3.,
       4., 2., 8., 0., 4., 7., 4., 2., 0., 0., 0., 7., 3., 0., 1., 2., 5.,
       3., 4., 4., 1., 2., 3., 9., 9., 3., 7., 0., 8., 8., 5., 4., 0., 9.,
       6., 0., 2., 6., 3., 4., 2., 4., 1., 2., 6., 9., 4., 7., 7., 8., 1.,
       6., 2., 3., 5., 7., 0., 7., 9., 2., 0., 9., 5., 4., 3., 0., 9., 0.,
       2., 6., 0., 1., 0., 1., 9., 8., 0., 6., 3., 5., 3., 6., 6., 4., 9.,
       7., 2., 1., 8., 7., 8., 7., 3., 3., 0., 6., 3., 9., 2., 1., 1., 2.,
       0., 4., 8., 8., 4., 0., 3., 2., 3., 4., 4., 5., 8., 7., 5., 9., 2.,
       0., 2., 4., 6., 2., 2., 1., 7., 1., 9., 4., 9., 5., 8., 8., 8., 2.,
       3., 9., 2., 5., 0., 5., 7., 1., 5., 7., 1., 7., 2., 8., 3., 4., 4.,
       1., 8., 7., 6., 8., 9., 6., 4., 4., 5., 4., 0., 4., 4., 2., 8., 2.,
       9., 8., 4., 9., 4., 0., 7., 1., 3., 5., 4., 9., 9., 8., 9., 2., 1.,
       7., 9., 2., 7., 5., 9., 6., 6., 6., 6., 6., 9., 9., 3., 6., 1., 5.,
       5., 0., 9., 0., 1., 6., 6., 2., 4., 3., 6., 4., 1., 0., 6., 7., 4.,
       9., 9., 1., 5., 0., 5., 3., 4., 2., 8., 7., 8., 1., 4., 5., 2., 0.,
       1., 4., 1., 5., 4., 1., 3., 1., 7., 6., 7., 5., 7., 7., 0., 5., 0.,
       9., 0., 9., 0., 4., 4., 5., 7., 7., 3., 9., 2., 4., 3., 1., 3., 1.,
       2., 3., 5., 4., 4., 9., 1.};`

image matrix
[... 287 lines cut ...]
          523., 528., 646., 453., 397., 463.],
         [519., 463., 421., 373., 602., 447., 440., 480., 526., 425.,
          527., 460., 579., 478., 494., 456.],
         [341., 312., 236., 237., 276., 262., 192., 326., 258., 222.,
          293., 335., 412., 249., 200., 244.]],

        [[348., 291., 338., 266., 448., 228., 298., 299., 359., 383.,
          320., 413., 387., 371., 414., 295.],
         [554., 564., 446., 434., 453., 426., 386., 490., 392., 470.,
          495., 524., 630., 457., 421., 428.],
         [596., 598., 465., 484., 646., 456., 434., 567., 464., 546.,
          520., 594., 676., 529., 575., 496.],
         [405., 404., 324., 281., 393., 359., 248., 403., 367., 326.,
          316., 427., 526., 443., 329., 338.]],

        [[416., 286., 221., 293., 334., 158., 250., 234., 207., 204.,
          276., 315., 289., 270., 220., 208.],
         [439., 370., 230., 308., 487., 389., 314., 358., 349., 373.,
          468., 461., 510., 423., 249., 289.],
         [467., 404., 290., 347., 507., 342., 265., 371., 286., 239.,
          369., 456., 461., 333., 292., 374.],
         [302., 258., 205., 234., 354., 287., 177., 251., 223., 252.,
          291., 392., 363., 262., 206., 338.]]]]`
IT IS CORRECT!!!!
I was shocked when I encounter this problem!

XiaochaoGONG, 2019-08-10:
Hi the different results may be caused by the padding method. In the first case, the output shape is 7x7, which means for input 14x14 and kernel 3x3 stride 2, there should be 1 additional zero padding. So the question is where to add the odd padding, if we add it at the left side, then the result is the same as PyTorch, while if we add it at the right side, the result is what you got. We refer the method of tensorflow [here](https://github.com/tensorflow/tensorflow/blob/3c3c0481ec087aca4fa875d6d936f19b31191fc1/tensorflow/core/framework/common_shape_fns.cc#L40-L48).
The second case needs 2 additional zero padding, which means one at the left side and one at the right side.

geekboood, 2019-08-10:
@XiaochaoGONG Thanks you for the reply! Now I understand it. However, could you add a support for this different padding strategy, such as a switch variable in the conv function which could be more convenient.