espressif/esp-who #108: Not working with OV5642 NODEMCU-32S Detected camera not supported.

weekeda, 2019-08-17:
Hi,

I have a Nodemcu-32S and an OV5642 I would love to use the software but after I upload the code I get following errors:

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

I don't know what to do now? Is it possible with the OV5642 asking google doesn't help to answer my question. I really would appreciate getting this camera working with the esp-who software. If its not possible i will buy a ESP-EYE but why buy new when I have already a more expensive camera lying around?

Thank you in advance for your help
Weekeda

bkeevil, 2019-08-20:
You will probably have better luck getting a response if you opened this issue in the esp32-camera repository. 

Someone would need to create a sensor driver for the OV5642. Here is where the code is for the sensors: https://github.com/espressif/esp32-camera/tree/master/sensors

weekeda, 2019-08-21:
Thank you @bkeevil ! 
I opened a second issue in the esp32-camera repository. 
Maybe someone with better understanding can help me to create a sensor driver for the OV5642 or supply a few hints...

bkeevil, 2019-08-22:
I came across this recently in the Linux kernel repository. I2S camera drivers for the Linux OS. They have drivers for the OV5640, OV5645, and OV5647. Lots of other camera drivers too. Maybe they can be used as a reference? I haven't even looked at them.

https://github.com/torvalds/linux/tree/master/drivers/media/i2c