espressif/esp-who #109: failed to build camera_web_server

greeran, 2019-08-20:
i am new  with the esp32 and am trying to build the camera web server example and fail with this output. if you could tell me what am i missing
thanks

single_chip/camera_web_server$ idf.py build
Checking Python dependencies...
Python requirements from /home/rang/esp/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory /home/rang/esp/project/cam-who/esp-who/examples/single_chip/camera_web_server/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /home/rang/esp/project/cam-who/esp-who/examples/single_chip/camera_web_server"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/rang/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/rang/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/rang/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/rang/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/rang/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v0.9.2-16-g1dff0bb
-- Building ESP-IDF components for target esp32
Loading defaults file /home/rang/esp/project/cam-who/esp-who/examples/single_chip/camera_web_server/sdkconfig.defaults...
-- Found PythonInterp: /home/rang/.espressif/python_env/idf4.0_py2.7_env/bin/python (found version "2.7.15") 
-- Found Perl: /usr/bin/perl (found version "5.26.1") 
-- Adding linker script /home/rang/esp/project/cam-who/esp-who/examples/single_chip/camera_web_server/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/rang/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/rang/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/rang/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/rang/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/rang/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/rang/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/rang/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
CMake Error at /home/rang/esp/project/cam-who/esp-who/components/esp-face/CMakeLists.txt:19 (target_link_libraries):
  Cannot specify link libraries for target "esp-face" which is not built by
  this project.

-- Configuring incomplete, errors occurred!
See also "/home/rang/esp/project/cam-who/esp-who/examples/single_chip/camera_web_server/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

xgarb, 2019-08-22:
I have a tutorial on my blog for this: https://robotzero.one/esp32-camera-module/ something there might help.

XiaochaoGONG, 2019-08-29:
What's your esp-idf version? It should be v3.2.

greeran, 2019-09-02:
hi. sorry for not answering i was on vacation. i worked with xgarg tutorial and it still outputs the same error and i am working on version v4.0

XiaochaoGONG, 2019-09-04:
There should be some tiny differences between the build scripts of 3.2 and 4.0. Anyway, either to go back to 3.2, either to change the CMakeLists.txt here.
https://github.com/espressif/esp-face/blob/679b5340418bd9604c25735294dc536babca3db6/CMakeLists.txt#L19-L20
Change the `esp-face` to `${COMPONENT_LIB}`

Xantrim, 2019-09-13:
I have the same problem using IDF V4.1. CMakeList.txt inside components/esp-face already contains the following lines:

`target_link_libraries(${COMPONENT_NAME} "-L ${CMAKE_CURRENT_SOURCE_DIR}/lib")
target_link_libraries(${COMPONENT_NAME}
    dl_lib
    fd
    fr
    )`

Is there any way to make this work? I don't want to have to switch between different IDF versions just to work with different boards.

XiaochaoGONG, 2019-09-16:
Hi all, the idf v4.0 has not been officially released. As v3.3 is released, please update the repo and submodules.

energo5, 2019-09-17:
If i already have version 3.3 what the best way to update repo and submodules? What commands in terminal on macos?

XiaochaoGONG, 2019-09-17:
> If i already have version 3.3 what the best way to update repo and submodules? What commands in terminal on macos?

`git pull origin master`
`git submodule update`

energo5, 2019-09-18:
> > If i already have version 3.3 what the best way to update repo and submodules? What commands in terminal on macos?
> 
> `git pull origin master`
> `git submodule update`

After this commands i see new error:
~/Documents/Arduino/libraries/esp-who/examples/single_chip/detection_with_command_line $ idf.py build
  File "/Users/shev/esp/esp-idf/tools/idf.py", line 436
    <<<<<<< HEAD
     ^
SyntaxError: invalid syntax

greeran, 2019-09-22:
hi
still not able to build the camera_web_server example. i tried pulling the git example but still fails on the same esp-face link library. i tried all changing to COMPONENT_NAME  but it doesnt help 
i started working with the ardoino ide but would prefer using the command line and understanding why it fails

XiaochaoGONG, 2019-09-23:
> hi
> still not able to build the camera_web_server example. i tried pulling the git example but still fails on the same esp-face link library. i tried all changing to COMPONENT_NAME but it doesnt help [...]

Hi greeran, I notice that you are using idf v4.0, which is not released yet, so we have not tested on that version. I've changed COMPONENT_NAME to COMPONENT_TARGET for idf v3.3, so you can test with that modification.

greeran, 2019-09-23:
thanks
changing to COMPONENT_TARGET solved the problem. now i have an undifined error but i will try and fix it

XiaochaoGONG, 2019-09-27:
> After this commands i see new error:
> ~/Documents/Arduino/libraries/esp-who/examples/single_chip/detection_with_command_line $ idf.py build
> File "/Users/shev/esp/esp-idf/tools/idf.py", line 436 [...]

Hi, it looks like you have modified something so that there are conflicts. So you need to fix the conflicts.
And you need to point esp-who idf to the right place. It seems you have already had a version of esp-idf. So either you update your esp-idf which seems to be /Users/shev/esp/esp-idf/, or modify your IDF_PATH to esp-who/esp-idf.

energo5, 2019-09-30:
>modify your IDF_PATH to esp-who/esp-idf.

It worked for me. Thanks.

richardgotthard, 2023-03-29:
I managed to get linking error: collect2: error:"dl_lib", "fd" and "fr" returned 1 exit status. Im mainly interested in getting dl_matrix3d to work. How should i best write target_link_libraries to make it work?

set(COMPONENT_SRCS
    face_detection/fd_forward.c
    face_recognition/fr_forward.c
    face_recognition/fr_flash.c
    image_util/image_util.c
    )

set(COMPONENT_ADD_INCLUDEDIRS
    face_detection/include
    face_recognition/include
    image_util/include
    lib/include
    )

set(COMPONENT_REQUIRES spi_flash)

register_component()

target_link_libraries(${COMPONENT_NAME} "-L ${CMAKE_CURRENT_SOURCE_DIR}/lib")
target_link_libraries(${COMPONENT_NAME}
    dl_lib
    fd
    fr
    )