espressif/esp-who #110: Motion Detection

billyrudi, 2019-08-26:
Hi, is it possible to detect motion only using video not pir or external sensor?
How i can detect if the frame video is changing ?
regards Paolo

mitchins, 2019-09-04:
This is an interesting application. You can definitely treat a face being detected as a sort of motion. Is there a way to run a higher level function such as Fourier to detect changes in the vector?

XiaochaoGONG, 2019-09-04:
Hi there are some APIs that would work for motion detection.

https://github.com/espressif/esp-face/blob/679b5340418bd9604c25735294dc536babca3db6/image_util/include/image_util.h#L305-L307

mitchins, 2019-09-04:
Interesting.
You’d probably want to create some sort of convolution so that a neighbourhood of a large size is considered.
Could work like:
1. Take image diff
2. Calculate neighbourhood sums that meet the threshold, and if a large enough motion is detected

billyrudi, 2019-09-05:
Hi ...but can you provide an example on ow do it?
i'm not so experienced on this library.