espressif/esp-who #111: how to convert my own trained model to your framework format?

hanson-young, 2019-08-29:
how to convert my own trained model to your framework format?

XiaochaoGONG, 2019-09-09:
If you have a trained model in .npy format, then you need to convert it to `dl_matrix3d_t` or `dl_matrix3dq_t` type. Please refer some detailed format [Here](https://github.com/espressif/esp-face/blob/master/lib/README.md)

wwzh2015, 2021-11-27:
> If you have a trained model in .npy format, then you need to convert it to `dl_matrix3d_t` or `dl_matrix3dq_t` type. Please refer some detailed format [Here](https://github.com/espressif/esp-face/blob/master/lib/README.md)

404

TiramisuJ, 2021-11-29:
@wwzh2015 Please refer [this new link](https://github.com/espressif/esp-dl/tree/master/tutorial)

closed 2022-09-21 by Auroragan