espressif/esp-who #112: Build error

energo5, 2019-09-06:
Hello.
When i follow this steps https://github.com/espressif/esp-who  on this step:
idf.py build

i see error:
CMake Error at /Users/shev/Documents/Arduino/libraries/esp-who/components/esp-face/CMakeLists.txt:19 (target_link_libraries):
  Cannot specify link libraries for target "esp-face" which is not built by
  this project.

Have you decision?

XiaochaoGONG, 2019-09-09:
You may have different version of esp-idf which we recommend. Anyway, please update the new version to 6d06fbb

energo5, 2019-09-10:
I have this version:
~/esp/esp-idf $ git describe --tags --dirty
v3.3

With new version CMakeLists.txt with this code:
register_component()

target_link_libraries(${COMPONENT_NAME} "-L ${CMAKE_CURRENT_SOURCE_DIR}/lib")
target_link_libraries(${COMPONENT_NAME}
    dl_lib
    fd
    fr
    )

I have the same error while "idf.py build"

XiaochaoGONG, 2019-09-16:
Please see #109

energo5, 2019-09-27:
Have you any decisions about this https://github.com/espressif/esp-who/issues/109#issuecomment-532508865 ?

closed 2019-09-30 by energo5