espressif/esp-who #113: ESP EYE Getting started guide lost information

magicseth, 2019-09-09:
https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP-EYE_Getting_Started_Guide.md

used to have valuable information on how to get started, but it was deleted!

XiaochaoGONG, 2019-09-16:
Hi, the software flow is moved [here](https://github.com/espressif/esp-who/blob/master/examples/single_chip/recognition_solution/README.md). Because ESP-EYE is not only for the example, so the information except hardware part is moved to example document .

davidallenmann, 2019-09-22:
The information is available on the Chinese language page, but missing from the English page.
https://github.com/espressif/esp-who/blob/master/docs/zh_CN/get-started/ESP-EYE_Getting_Started_Guide.md

closed 2019-09-30 by XiaochaoGONG