espressif/esp-who #114: How to calibrate the camera in ESP32

hughhugh, 2019-09-14:
Hi. When I am using ESP32 to implement the qr or other image processing methods, the camera is generally calibated, then how can I calibrate the esp32 camera or utilizing the camera infos in esp32? Thank you.

me-no-dev, 2019-09-14:
What do you mean by "calibrate" and "camera infos"? Have a look at the web server example. It shows everything that you can do with the camera :)

hughhugh, 2019-09-14:
I have read and search the wiki. I donot find the camera calibartion document. The camera calibration is just  like this site.
https://blog.csdn.net/weixin_43331257/article/details/82932904
https://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html

xenpac, 2020-03-13:
There is NO camera calibration/Lens correction in the software.
That would also depend on the camera modul used.
Lens correction (if any) is integrated in the camera modul itself.
Also i havent seen opencv in here.