espressif/esp-who #115: This URI doesn't exist

energo5, 2019-09-30:
Hello.
I connect to esp_eye and when i go to http://192.168.4.1/face_stream in browser i see:
This URI doesn't exist

This error in monitor:
W (324614) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist

Can you help me?

me-no-dev, 2019-09-30:
which example? on the web server it's http://192.168.4.1:81/stream

energo5, 2019-09-30:
This example:
https://github.com/espressif/esp-who/tree/master/examples/single_chip/recognition_solution

http://192.168.4.1:81/stream also not working

XiaochaoGONG, 2019-09-30:
This example is dedicated for ESP-EYE with its specific functions, which are described in the README.md.

energo5, 2019-09-30:
I specifically bought esp-eye and try to run on it.

energo5, 2019-10-03:
Hello.
I try further work with ESP-EYE with this example https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server
Its normal build, flash.
In monitor i see:
I (1664) wifi: connected with WiFi110, channel 10
I (1674) wifi: pm start, type: 1
I (3204) event: sta ip: 192.168.1.69, mask: 255.255.255.0, gw: 192.168.1.1
I (3204) camera wifi: got ip:192.168.1.69
 
Browser print "cannot connect to server" when i goto 192.168.1.69.

After some time monitor print:
E (129444) camera: Detected camera not supported.
E (129444) camera: Camera probe failed with error 0x20004
E (129444) app_camera: Camera init failed with error 0x20004
I (129444) camera_httpd: Starting web server on port: '80'
I (129454) camera_httpd: Starting stream server on port: '81'

Further i try this example https://github.com/espressif/esp-who/tree/master/examples/single_chip/detection_with_command_line
Its normal build and flash.
In monitor i see this last log lines:
I (0) cpu_start: App cpu up.
I (751) spiram: SPI SRAM memory test OK
I (751) heap_init: Initializing. RAM available for dynamic allocation:
I (751) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (757) heap_init: At 3FFB4460 len 0002BBA0 (174 KiB): DRAM
I (764) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (770) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (776) heap_init: At 40090E98 len 0000F168 (60 KiB): IRAM
I (783) cpu_start: Pro cpu start user code
I (787) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (136) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (137) sccb: pin_sda 26 pin_scl 27

and nothing else not happening when i see at camera with difference distance.
Then after 2-3 minutes monitor print this:
E (129157) camera: Detected camera not supported.
E (129157) camera: Camera probe failed with error 0x20004
E (129157) app_camera: Camera init failed with error 0x20004
E (129157) app_process: Camera capture failed
E (129167) app_process: Camera capture failed
E (129167) app_process: Camera capture failed
E (129177) app_process: Camera capture failed
E (129177) app_process: Camera capture failed
E (129187) app_process: Camera capture failed
E (129187) app_process: Camera capture failed
E (129197) app_process: Camera capture failed
E (129197) app_process: Camera capture failed
E (129207) app_process: Camera capture failed

The i try this example https://github.com/espressif/esp-who/tree/master/examples/single_chip/recognition_with_command_line
Its normal buld and flash.
In monitor i see last log lines:
I (0) cpu_start: App cpu up.
I (1054) spiram: SPI SRAM memory test OK
I (1055) heap_init: Initializing. RAM available for dynamic allocation:
I (1055) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1061) heap_init: At 3FFB45A0 len 0002BA60 (174 KiB): DRAM
I (1068) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1074) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1081) heap_init: At 40090E98 len 0000F168 (60 KiB): IRAM
I (1087) cpu_start: Pro cpu start user code
I (1092) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (217) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (218) sccb: pin_sda 26 pin_scl 27

and nothing happened when i see at ESP-EYE at difference distance.
After 2 minutes monitor print:

E (129238) camera: Detected camera not supported.
E (129238) camera: Camera probe failed with error 0x20004
E (129238) app_camera: Camera init failed with error 0x20004
E (129238) app_process: Camera capture failed
E (129248) app_process: Camera capture failed
E (129248) app_process: Camera capture failed
E (129258) app_process: Camera capture failed
E (129258) app_process: Camera capture failed
E (129268) app_process: Camera capture failed

What you can say? Its mean that ESP-EYE not working and i must return ESP-EYE to seller?

XiaochaoGONG, 2019-10-03:
Hi, for all the undetected message, did you check your configuration that maps the camera pins to the right device ? You can run `idf.py menuconfig` then find the ESP-EYE mapping. By default, the mapping is for WROVER-Kit.

energo5, 2019-10-03:
I checked camera pins.
In this examples "detection_with_command_line" and "recognition_with_command_line" i see that WROVER-KIT already selected:
[image]
[image]

XiaochaoGONG, 2019-10-03:
So you need to change to ESP-EYE pin, entering the `Select Camera Pinout`, there is an option for it.

energo5, 2019-10-04:
After select pins for ESP-EYE this examples "camera_web_server", "detection_with_command_line" and "recognition_with_command_line" started work right.
But example "recognition_solution" still no work.
I can not find CAMERA PINS in menuconfig:
[image]
How set pins in this example?

energo5, 2019-10-09:
Hello.
Can you help me with this error - https://github.com/espressif/esp-who/issues/115#issue-500048485 in example "recognition_solution"?

XiaochaoGONG, 2019-10-09:
Hi @energo5  please follow the [Key-process](https://github.com/espressif/esp-who/tree/master/examples/single_chip/recognition_solution#key-process).
The basic idea is to firstly awake the device, then to connect to the network, then to do face detection.

energo5, 2019-10-10:
Hello
The problem was that I didn’t turn on the downloaded greeting recording loudly.
It did not work, and I commented on these lines in the code to skip waiting for a greeting:
     while (g_state == WAIT_FOR_WAKEUP)
         vTaskDelay (1000 / portTICK_PERIOD_MS);
After that I got the error https://github.com/espressif/esp-who/issues/115#issue-500048485

Now I turn on the greeting record loudly and everything is working.
Tell me how to bypass the waiting code in the code?

XiaochaoGONG, 2019-10-10:
Change the `WAIT_FOR_WAKEUP` to `WAIT_FOR_CONNECT`
https://github.com/espressif/esp-who/blob/19123aaeae5f0ac1fff434df5ee65a823b6fabea/examples/single_chip/recognition_solution/main/app_main.c#L99
and 
https://github.com/espressif/esp-who/blob/19123aaeae5f0ac1fff434df5ee65a823b6fabea/examples/single_chip/recognition_solution/main/app_httpserver.c#L347

energo5, 2019-10-11:
Its worked.
Fine.
Thank you very much.

closed 2019-10-11 by energo5