espressif/esp-who #116: Problem building camera_web_server

fdoljack, 2019-10-03:
This one is perplexing.  I am using esp-idf version 4.1.  esp-who and the components esp32-camera and esp-face are the current commits.  The camera is an ESP-cam AIThinker.  Below I post the build output.  The fix may be something that is obvious that I am missing:
Executing action: all (aliases: build)
Running make in directory /home/frank/esp2/esp-who/examples/single_chip/camera_web_server/build
Executing "make -j 6 all"...
-- Project version: v0.9.3-1-g19123aa
-- Building ESP-IDF components for target esp32
Loading defaults file /home/frank/esp2/esp-who/examples/single_chip/camera_web_server/sdkconfig.defaults...
-- Adding linker script /home/frank/esp2/esp-who/examples/single_chip/camera_web_server/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/frank/esp2/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/frank/esp2/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/frank/esp2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/frank/esp2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/frank/esp2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/frank/esp2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-face esp-tls esp32 esp32-camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs fb_gfx freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread recorder_engine sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/frank/esp2/esp-idf/components/app_trace /home/frank/esp2/esp-idf/components/app_update /home/frank/esp2/esp-idf/components/asio /home/frank/esp2/esp-idf/components/bootloader /home/frank/esp2/esp-idf/components/bootloader_support /home/frank/esp2/esp-idf/components/bt /home/frank/esp2/esp-idf/components/cbor /home/frank/esp2/esp-idf/components/coap /home/frank/esp2/esp-idf/components/console /home/frank/esp2/esp-idf/components/cxx /home/frank/esp2/esp-idf/components/driver /home/frank/esp2/esp-idf/components/efuse /home/frank/esp2/esp-who/components/esp-face /home/frank/esp2/esp-idf/components/esp-tls /home/frank/esp2/esp-idf/components/esp32 /home/frank/esp2/esp-who/components/esp32-camera /home/frank/esp2/esp-idf/components/esp_adc_cal /home/frank/esp2/esp-idf/components/esp_common /home/frank/esp2/esp-idf/components/esp_eth /home/frank/esp2/esp-idf/components/esp_event /home/frank/esp2/esp-idf/components/esp_gdbstub /home/frank/esp2/esp-idf/components/esp_http_client /home/frank/esp2/esp-idf/components/esp_http_server /home/frank/esp2/esp-idf/components/esp_https_ota /home/frank/esp2/esp-idf/components/esp_https_server /home/frank/esp2/esp-idf/components/esp_local_ctrl /home/frank/esp2/esp-idf/components/esp_ringbuf /home/frank/esp2/esp-idf/components/esp_rom /home/frank/esp2/esp-idf/components/esp_websocket_client /home/frank/esp2/esp-idf/components/esp_wifi /home/frank/esp2/esp-idf/components/espcoredump /home/frank/esp2/esp-idf/components/esptool_py /home/frank/esp2/esp-idf/components/expat /home/frank/esp2/esp-idf/components/fatfs /home/frank/esp2/esp-who/components/fb_gfx /home/frank/esp2/esp-idf/components/freemodbus /home/frank/esp2/esp-idf/components/freertos /home/frank/esp2/esp-idf/components/heap /home/frank/esp2/esp-idf/components/idf_test /home/frank/esp2/esp-idf/components/jsmn /home/frank/esp2/esp-idf/components/json /home/frank/esp2/esp-idf/components/libsodium /home/frank/esp2/esp-idf/components/log /home/frank/esp2/esp-idf/components/lwip /home/frank/esp2/esp-who/examples/single_chip/camera_web_server/main /home/frank/esp2/esp-idf/components/mbedtls /home/frank/esp2/esp-idf/components/mdns /home/frank/esp2/esp-idf/components/mqtt /home/frank/esp2/esp-idf/components/newlib /home/frank/esp2/esp-idf/components/nghttp /home/frank/esp2/esp-idf/components/nvs_flash /home/frank/esp2/esp-idf/components/openssl /home/frank/esp2/esp-idf/components/partition_table /home/frank/esp2/esp-idf/components/protobuf-c /home/frank/esp2/esp-idf/components/protocomm /home/frank/esp2/esp-idf/components/pthread /home/frank/esp2/esp-who/components/recorder_engine /home/frank/esp2/esp-idf/components/sdmmc /home/frank/esp2/esp-idf/components/soc /home/frank/esp2/esp-idf/components/spi_flash /home/frank/esp2/esp-idf/components/spiffs /home/frank/esp2/esp-idf/components/tcp_transport /home/frank/esp2/esp-idf/components/tcpip_adapter /home/frank/esp2/esp-idf/components/ulp /home/frank/esp2/esp-idf/components/unity /home/frank/esp2/esp-idf/components/vfs /home/frank/esp2/esp-idf/components/wear_levelling /home/frank/esp2/esp-idf/components/wifi_provisioning /home/frank/esp2/esp-idf/components/wpa_supplicant /home/frank/esp2/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/frank/esp2/esp-who/examples/single_chip/camera_web_server/build
Scanning dependencies of target esp32_linker_script
Scanning dependencies of target _project_elf_src
Scanning dependencies of target partition_table
[  0%] Generating project_elf_src.c
[  1%] Generating ../../partition_table/partition-table.bin
[  1%] Generating linker script...
[  1%] Built target _project_elf_src
[  1%] Built target partition_table
Scanning dependencies of target bootloader
[  1%] Creating directories for 'bootloader'
[  1%] Built target esp32_linker_script
Scanning dependencies of target __idf_tcpip_adapter
[  1%] No download step for 'bootloader'
[  1%] No update step for 'bootloader'
[  1%] No patch step for 'bootloader'
[  1%] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.7.4") 
[  2%] Building C object esp-idf/tcpip_adapter/CMakeFiles/__idf_tcpip_adapter.dir/event_handlers.c.obj
[  2%] Building C object esp-idf/tcpip_adapter/CMakeFiles/__idf_tcpip_adapter.dir/tcpip_adapter_lwip.c.obj
[  2%] Linking CXX static library libtcpip_adapter.a
[  2%] Built target __idf_tcpip_adapter
Scanning dependencies of target __idf_lwip
[  2%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/sntp/sntp.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/dhcpserver/dhcpserver.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/api_lib.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/ping/ping.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/ping/esp_ping.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/api_msg.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/err.c.obj
[  3%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/netbuf.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/netdb.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/netifapi.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/sockets.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/api/tcpip.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/apps/sntp/sntp.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/def.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/dns.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/inet_chksum.c.obj
[  4%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/init.c.obj
[  5%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/ip.c.obj
[  5%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/mem.c.obj
[  5%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/memp.c.obj
[  5%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/netif.c.obj
[  5%] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/pbuf.c.obj
[... cut ...] efined reference to `dl_matrix3dqq_conv_1x1_with_bias'
collect2: error: ld returned 1 exit status
CMakeFiles/camera_web_server.elf.dir/build.make:263: recipe for target 'camera_web_server.elf' failed
make[2]: *** [camera_web_server.elf] Error 1
CMakeFiles/Makefile2:501: recipe for target 'CMakeFiles/camera_web_server.elf.dir/all' failed
make[1]: *** [CMakeFiles/camera_web_server.elf.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

closed 2019-10-04 by me-no-dev

fdoljack, 2019-10-04:
After using idf v3.2 it worked like a charm

nissimzur, 2020-07-07:
This is a very bad soultion. since we like to progress with new in IDF. 
Is there any other way to fix that error with IDF 4.2?

pushtoopen, 2020-10-09:
FYI. This error happens in 4.2 likely because you moved from a different branch (git checkout.) Do a git status and remove the modified folders then git reset --hard to get the latest. Don't forget to reinstall your idf using install.sh in the esp-idf folder.