espressif/esp-who #117: ESP-EYE: is it possible to activate the device as iBeacon?

krambriw, 2019-10-04:
So I have a number of ESP-EYE's running since long, working very well using the camera_web_server example. Now I was just thinking, could it be possible to also make them act as iBeacons?
I have seen there are a number of bluetooth examples in the esp-idf like ble_adv and ble_ibeacon

But how would you add such an example to the camera_web_server project to build and flash both to the esp32?

XiaochaoGONG, 2019-10-09:
I've not tested that example, but I might give you an advice. 
1. Copy the files from idf/examples/bluetooth/ble_ibeacon/main to esp-who/components/ble_ibeacon (need to mkdir ble_ibeacon)
2. Rename the `app_main` in the ibeacon_demo.c with `ibeacon_start (actually whatever name)`
3. Call the function in the real app_main in camera_web_server

krambriw, 2019-10-09:
Thank you for helping me out, I have followed your advice. When I do make -j5 I get this error message:

CC build/main/app_main.o
C:/msys32/home/EG/esp/esp-who/examples/single_chip/camera_web_server/main/app_main.c: In function 'app_main':
C:/msys32/home/EG/esp/esp-who/examples/single_chip/camera_web_server/main/app_main.c:32:5: error: implicit declaration of function 'ibeacon_start' [-Werror=implicit-function-declaration]
     ibeacon_start();
     ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/EG/esp/esp-who/esp-idf/make/component_wrapper.mk:286: app_main.o] Fel 1
make: *** [C:/msys32/home/EG/esp/esp-who/esp-idf/make/project.mk:505: component-main-build] Fel 2

krambriw, 2019-10-09:
My app_main.c in the camera_web_server

#include "app_camera.h"
#include "app_wifi.h"
#include "app_httpd.h"

void app_main()
{
    app_wifi_main();
    app_camera_main();
    app_httpd_main();
    ibeacon_start();
}

XiaochaoGONG, 2019-10-10:
missing the declaration of `ibeacon_start`, put `void ibeacon_start();` above.

krambriw, 2019-10-10:
Yes, I figured out, thanks. It builds correctly without errors during make and the result can be flashed. But the ibeacon doesn't start. Also tried just with ibeacon but no success

```
void app_main()
{
/*app_wifi_main();
app_camera_main();
app_httpd_main();*/
ibeacon_start();
} 
```
I then tried to build the sample in esp-idf itself, flashed it to the esp-eye and the ibeacon started! So means the bluetooth in the esp-eye is functioning correctly

krambriw, 2019-10-10:
I tried like this of course
```
void app_main()
{
/*app_wifi_main();
app_camera_main();
app_httpd_main();*/
void ibeacon_start();
} 
```

XiaochaoGONG, 2019-10-10:
Please check your sdkconfig file to see if there is IBEACON_MODE like macros.

krambriw, 2019-10-10:
Yes, it looks ok if I compare with the sdkconfig in the esp-idf ble_ibeacon project
Something is missing...

[sdkconfig.txt](https://github.com/espressif/esp-who/files/3711012/sdkconfig.txt)

krambriw, 2019-10-11:
After many attempts, I just once saw "register callback" in the monitor log from the "IBEACON_DEMO" module. But then everything else did not continue, the camera did not work, the wifi did not connect, the web server did not start. So a kind of blocking happened
I could not re-create this anymore. But it showed that the function call at least is happening but for some reason it does not work, maybe a memory problem, maybe is asking too much from the esp-eye
Attached is the normal result from monitor when camera is working, and ibeacon is not. There is a warning about window overflow
[make_monitor.txt](https://github.com/espressif/esp-who/files/3715928/make_monitor.txt)

krambriw, 2019-10-18:
Thanks for looking into this
I made those changes, allocate also just 1 frame but sorry, no difference, still not working

krambriw, 2019-10-27:
seems this  is not possible

closed 2019-10-27 by krambriw