espressif/esp-who #119: SPI port

Skyseb, 2019-10-11:
Hi,

In:

https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP-EYE_Getting_Started_Guide.md

says.. A reserved port for data transmission.

CS I/O 19  PART OF VSPI
D0  I/O 12 HSD2 ... (for sd in mmc mode)
DI   I/O 2 PART OF HSPI
CLK I/O 0  CLK

what does it mean?

i cant use it ?

otherwise:

can i use it to put an Micro SD card in SPI mode?
I NEED IT

can i use it for I/O?
I settle for that

XiaochaoGONG, 2019-10-15:
Hi, the pads in the board are for SPI usage. I think you need to connect the IOs with your own jump wire.

feklee, 2019-10-27:
I tried it. As soon as I connect CLK from another ESP32, which acts as host, I cannot monitor output on the ESP-EYE anymore.

Can someone clarify: *Can this SPI port really be used for communication with other devices?*

feklee, 2019-10-27:
Looking at the schematic, I see as connections:

  * `CLK` = `IO0`:

      - `GPIO0`

      - Input for the `download` button. (which probably is the button labeled *BOOT* on the board)

      - Connected to `BC847BDW1T1G` in the circuit *USB & USB=>UART.*

  * `DOUT` = `IO12`:

      - `GPIO12`

      - pulled high via a 10k resistor connected to 3.3V

  * `CS` = `IO19`, `DIN` = `IO2`: connected to `GPIO19` and `GPIO2` respectively

feklee, 2019-10-28:
Got it to work (Arduino Pro Mini as master, ESP-EYE as slave):

[image]

Note that the ESP-EYE will not boot if there is clock signal on the `CLK` pin. According to my tests, the `CLK` pin needs to be unconnected during boot or pulled high. For communication, I use SPI in mode 1.

BRichter99, 2020-09-30:
> I tried it. As soon as I connect CLK from another ESP32, which acts as host, the ESP-EYE does not boot anymore. When I monitor the `hello_world` example, it hangs at: `waiting for download`
Where can I find the code of that example?

> Got it to work (Arduino Pro Mini as master, ESP-EYE as slave):
Which code did you use on the ESP-EYE? Was it that hello_world example?

feklee, 2020-09-30:
[code.zip](https://github.com/espressif/esp-who/files/5306123/code.zip)

*This and the image from my earlier post go together with my notes below. I used this method in my project [SAN](https://github.com/feklee/san).*

Warning: Note that the voltage divider circuit is not wired correctly.  There is 5 V on the ESP32 pins.

Relationship:

  * Arduino: master

  * ESP-EYE: slave

The master sends to the slave (no exclamation mark):

    Hello, world

Components:

  * Arduino Pro Mini

  * ESP-EYE

  * voltage dividers, each consisting of a 3300 Ω resistor (connected to ground) and a 1700 Ω resistor (created from three smaller ones)

Process to set up the ESP-EYE (ESP-WHO v3.3):

  * Configure:

        $ python2 $(which idf.py) menuconfig

  * Build:

        $ python2 $(which idf.py) build

  * Flash:

        $ python2 $(which idf.py) -p /dev/ttyUSB0 flash

    When the Arduino is powered, the ESP-EYE cannot be flashed.

  * To view serial monitor:

        $ python2 $(which idf.py) -p /dev/ttyUSB0 monitor

    When the Arduino there is a signal on the clock line or if it is pulled low, the ESP-EYE won’t boot and cannot be updated. Therefore the Arduino delays SPI communication when it starts. So simply reset the Arduino, wait until the pin 13 is high (green LED), then boot the ESP-EYE.

    It can take a while until “Hello, world” appears.

With DMA enabled, the length of the transmitted message needs to be multiple of 32 bits.  SPI mode is set to 1 per recommendation of the ESP32 SPI/DMA documentation.