espressif/esp-who #122: matrix3du item alloc faield

Mohammed-Ka, 2019-10-26:
Hi,

Recently i purchased this board: TTGO T-JOURNAL ESP32 CAM. I'm trying to run the ESP-WHO default CameraWebserver example on it, everything worked fine except for face recognition and detection. Everytime i press the face recognition button the camera turns black and i get this error in serial monitor :

matrix3du item alloc faield 

dl_matrix3du_alloc  

I did some research about it and find out the problem is from esp32 board core so i downgraded to version 1.0.1 as recommended and tried again and still same error. Any suggestions to solve this issue would be highly appreciated.

Thank you.

[image]

[image]
[image]

me-no-dev, 2019-10-27:
did you enable psram in menuconfig?

Mohammed-Ka, 2019-10-27:
Yes i enabled it and i tried to flash the code twice one with esp-idf and one with arduino ide. Same problem everytime, I've been trying for almost one week now without solution

XiaochaoGONG, 2019-10-29:
Can you post on your sdkconfig and some logs at the initial steps ?

Mohammed-Ka, 2019-11-20:
Unfortunately this model don't have psram, i asked the manufacturer about it. Thanks guys

closed 2019-11-20 by Mohammed-Ka