espressif/esp-who #124: Program ESP32-CAM via FTDI without manual reset?

feklee, 2019-11-15:
When programming the ESP32-CAM with an FTDI, I proceed as follows:

 1. Connect IO0 to GND.

 2. Reset the ESP32-CAM. (by cutting power)

 3. Run:

        python2 $(which idf.py) -p /dev/ttyUSB0 flash --baud 2000000

Afterwards, to monitor output:

 1. Remove IO0 from GND.

 2. Reset the ESP32-CAM. (as above)

 3. Run:

        python2 $(which idf.py) -p /dev/ttyUSB0 monitor

*Is there a way to do all that without touching any cables or buttons?*

xenpac, 2019-12-09:
No.
The Jumper is needed to enter programming mode.

xenpac, 2019-12-10:
ps. i just thought that on the serial adapter you have the RTS and DTR signals that can be used to drive 
the programming pin. However there is no external RESET pin on the esp32-cam board .
So....go manual.

feklee, 2019-12-11:
Thanks for the suggestions! I also considered the using the RTS and DTR pins on the serial adapter. In the end I did use an ESP-EYE (for different reasons).

closed 2019-12-11 by feklee