espressif/esp-who #125: Unable to program ESP-EYE Dev Board V2.1, Timed out waiting for packet header

dheerajdake, 2019-12-07:
Hello all,
I'm struggling to program this board. I've got all the setup required. My environment is Ubuntu in Virtualbox. Network and USB ports have been mapped. I'm able to see the ESP EYE serial output.

For this test, I'm using the hello_world app from this [page](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-connect). I was able to program the application initially using

> idf.py -p /dev/ttyUSB0 -b 115200 flash

After this I was able to see the example output using putty.

However, I was not able to re-flash the board at all. The above command results in this output.

> Adding "flash"'s dependency "all" to list of commands with default set of options.
> Executing action: all (aliases: build)
> Running ninja in directory /home/dheeraj/esp/hello_world/build
> Executing "ninja all"...
> [1/3] Performing build step for 'bootloader'
> ninja: no work to do.
> Executing action: flash
> Running esptool.py in directory /home/dheeraj/esp/hello_world/build
> Executing "/home/dheeraj/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/dheeraj/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
> esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
> esptool.py v2.9-dev
> Serial port /dev/ttyUSB0
> Connecting........_____....._____....._____....._____....._____....._____....._____
> 
> A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
> esptool.py failed with exit code 2

ESP-EYE board is connected using the original USB cable to the computers USB 2.0 port. Any help would be appreciated.

Thanks
Dheeraj

dheerajdake, 2019-12-07:
I tried this considering my previous experience with ESP8266.
1. Held reset button
2. Held boot button while reset is still held
3. Released reset button while boot button is still held

The output in the serial changed as follows:
[image]

I tried to program at this stage with flash command but the flash command used to reset the device back to flash mode. So I tried doing the above sequence during Connecting.... stage. From the serial output, it looked like something was being transmitted when the ESP was in waiting for download stage. But in the end programming failed.
[image]

XiaochaoGONG, 2019-12-10:
Your serial port seems to be occupied by putty while flashing, you need to disconnect from putty then to flash.