espressif/esp-who #127: Esp-Eye build problem (IDFGH-2487) (IDFGH-2488)

lukasz139, 2020-01-08:
Hi,

I am trying to build an example project for the Esp-Eye board. I have installed esp-idf and set up the environmental variables. I am able to build and test the example projects inside of the esp-idf/examples directory. I know that library components inside of the esp-who are necessary to use Esp-Eye, so I set up the esp-who directory as well. Unfortunately, the "detection_with_command_line" example project fails the build process.

I get the following error message:

`CMake Error at C:/Users/lhuch/esp-who/esp-idf/tools/cmake/component_utils.cmake:20 (message):
  Required component esp32-camera is not found in any of the provided
  COMPONENT_DIRS
Call Stack (most recent call first):
  C:/Users/lhuch/esp-who/esp-idf/tools/cmake/scripts/expand_requirements.cmake:95 (find_component_path)
  C:/Users/lhuch/esp-who/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  C:/Users/lhuch/esp-who/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at C:/Users/lhuch/esp-who/esp-idf/CMakeLists.txt:39 (message):
  Failed to expand component requirements

-- Configuring incomplete, errors occurred!
See also "C:/Users/lhuch/EspWhoWorkspace/sol/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/lhuch/EspWhoWorkspace/sol/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1`

I have tried adding the component directory of esp-who to the EXTRA_COMPONENT_DIRS variable but without success. Any ideas?

Thanks in advance.

projectgus, 2020-01-08:
Hi @lukasz139 ,

I've moved this to the esp-who project as it seems to be related to that.

What IDF version are you using? ESP-WHO recommends v3.3.1, if you have the latest IDF master branch then it is unlikely to work.

XiaochaoGONG, 2020-01-09:
Hi @lukasz139, did you update the submodules of esp-who ? With `git submodule update --init` you can get all submodules needed. 
If you have already had esp-idf and don't want to redownload in esp-who, you can pick all submodule except esp-idf, e.g. `git submodule update components/esp32-camera`, etc.

zikalino, 2023-08-17:
Seems there's no activity here, I am closing this issue. Please reopen if you still need help with this problem.

tom-borcin, 2024-01-04:
I am closing this issue, since the issue is stale. Let me know if you still need help with this issue and we will reopen it.

closed 2024-01-04 by tom-borcin