espressif/esp-who #128: Can't build with Cmake.  Is it not supported?

mkeveney, 2020-01-16:
I've successfully built the camera_web_server example using the msys32 environment, but that's miserably slow under Windows.

When I try the Cmake (preview, since we need 3.3.1) I get build failure, since bootloader.bin ends up too large:

> esptool write_flash: error: argument <address> <filename>: Detected overlap at
> address: 0x8000 for file: partition_table/partition-table.bin

I suspect this has to do with configuration, as the resulting sdkconfig is quite different when using 'idf.py menuconfig' vs the old 'make menuconfig'

Any thoughts?

XiaochaoGONG, 2020-01-17:
Hi, it may be due to the enlargement of the bootloader in idf 3.3.1. Anyway, one solution is to modify the offset of partition table, e.g. to 0x9000.
[image]

BTW, the low speed in windows, do you mean in the flash process? If so, you need to change the flash baud rate higher, e.g. `idf.py flash -b 921600`

mkeveney, 2020-01-17:
> Hi, it may be due to the enlargement of the bootloader in idf 3.3.1

I'm using 3.3.1 in both cases.  The code is the same.
The difference is sdkconfig; different compile options are active, 
resulting in a very different bootloader.bin:

  using msys32, size is 27,696
  using Cmake, it's 29,456

I've accepted defaults and changed the same 3 things in each case.  
Shouldn't my sdkconfig files be identical?

>  Anyway, one solution is to modify the offset of partition table, e.g. to 0x9000.

I thought about that, but I'd really like to understand why they're different.. 
in case I'm missing something _else_ that might be more significant.

> BTW, the low speed in windows, do you mean in the flash process? 

No the actual flash process is about the same, once it finally gets there.  
It's the 'make' process itself that's afwul.  Running 'make' with no paramters 
after touching a single file takes over 30 seconds.  'idf.py build' takes 
about 4 seconds.  (That's without flashing at all.)

Thanks for your reply!

-Matt

mkeveney, 2020-01-17:
I've gone into this a bit deeper.  I no longer suspect the sdkconfig file difference.  They're arranged differently, but funcitonally they seem to be identical, except for the documented differences (no flash port defined in the Cmake variant).

So, I suspect something else is making the bootloader larger, but don't have details yet.  Will report them here when I do.

Meanwhile, if anyone else has run into this, please share your findings!

me-no-dev, 2020-01-19:
I wonder is cmake is building with slightly different flags and that to be causing the size difference. Have you seen size difference in the firmware bin also?

mkeveney, 2020-01-20:
The main .bin file _is_ slightly different, but by less than 1K.  I'm not sure why.  I did notice that some of the error messages incorporate the source path, which would account for minor differences in size.

I'm more concerned about the boot loader, but interested in both.