espressif/esp-who #130: ESP-EYE build face_detection example(detection with command line)-error

DanishSheikh1999, 2020-02-16:
/home/danish/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/danish/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
/home/danish/esp/esp-who/esp-idf/make/component_wrapper.mk:289: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 1
/home/danish/esp/esp-who/esp-idf/make/project.mk:552: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2

bitmakk, 2020-02-16:
Do you have the proper idf version?

shinmai, 2020-02-27:
> Do you have the proper idf version?

Is the "proper" IDF version still the 3.x version from 6 months ago or the actual real current stable release? Don't have the exact same issue as OP, but trying to figure out if it's worth the hassle to set up esp-who at this point or if I should just toss my esp-eye.

me-no-dev, 2020-02-28:
the camera driver supports idf 4+

DanishSheikh1999, 2020-03-04:
proper idf version solved the problem

closed 2020-03-04 by DanishSheikh1999