espressif/esp-who #132: Failed to build with IDF4.0 or newer

jiangzhiyong2020, 2020-03-04:
I need to implement wifi/camera function for a product and developed a board with ESP32 to evaluate. I followed latest version document (for idf 4.0 or master version) to install idf and use eclipse IDE to compile and debug with Jtag. Everything is fine. But when I downloaded ESP-eye  example for camera, it failed to build it with idf 4.0 or newer. I tried it last year with old MSYS version with esp-eye and it could compile. After reading the documentation about cmake building for new version IDF, I modified all CMakeLists.txt files in the project with new command  idf_component_register(..), and keep other content not changed. But it still have 39 errors, all of them similar to below:
/components/esp-face/lib\libfd.a(mtmn_lite_q.c.obj):(.literal.pnet_lite_q+0x60): undefined reference to `dl_matrix3duq_mobilenet'
...
/home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_detection/mtmn/mtmn_lite_q.c:90: undefined reference to `dl_matrix3dqq_mobilenet'

Do you have a project which is compatible with idf 4.0?  it will be best if Espressif can update ESP-who for latest idf. I have spent many days and still can not figure out how to do this. Thanks for help.

jiangzhiyong2020, 2020-03-04:
I just installed the idf Ver. 3.3.1 and the project builds fine. For now, I will just use older version idf.

bitmakk, 2020-03-04:
i have the same problem, because some "smart" guy hardcoded the full (now incorrect) include paths into the libraries. But I need the IDF4 because I use a lot of function from it to my project.
I hope I can bypass the error, or they will fix it

jiangzhiyong2020, 2020-03-05:
After I changed from IDF 4.0 to IDF3.3 so that I can use camera functions, many other functions, such as event, wifi, and uart are not working anymore and I have to change many places for IDF3.3. I really wish espressif can update esp-who project with IDF4.0 ASAP.

me-no-dev, 2020-03-05:
the camera driver IS IDF4 (both make and cmake) compatible already. If you do not need face detection/recognition, then you will be fine with just the camera driver (pull as any other component in your project).

@bitmakk the "smart" comment was unnecessary. We can not see in the future when we write something and vision what will happen to paths and headers. ESP32-S2 was only a project back then (the reason why headers moved)

Hazeline2018, 2020-03-06:
@bitmakk I encountered the same problem while using IDF v4.0. The example projects included couldn't even build successfully. Could you please tell me where and how to change in the make tool chain to have it work for IDF4.0? Thanks a bunch!!!

bitmakk, 2020-03-06:
I couldn't figure it out unfortunately :\

jiangzhiyong2020, 2020-05-05:
Today I downloaded the new esp-who master version. Finally, it works with the new IDF4.0 framework and I can keep using all the new functions/structures in IDF4.0 instead of old IDF3.3. What a relief :). Thank you, Expressif support team.

closed 2020-05-05 by jiangzhiyong2020