espressif/esp-who #140: Reflash - boot partition too large

eponymic, 2020-04-19:
I am trying to get back to the out-of-the-box functionality.  When I flash the camera_web_server project to the esp-eye it appears to work.  Then when I monitor I get the below error

E (898) esp_image: Image length 2659392 doesn't fit in partition length 1048576
E (898) boot: Factory app partition is not bootable
E (900) boot: No bootable app partitions in the partition table

My partiction.csv is as follows.
# Espressif ESP32 Partition Table
# Name,  Type, SubType, Offset,  Size
factory, app,  factory, 0x010000, 3840K
nvs,     data, nvs,     0x3D0000, 16K

chegewara, 2020-04-19:
Looks like your custom partition table is not used:
` partition length 1048576`
Did you change in menuconfig to use custom partition table csv file?

eponymic, 2020-04-20:
No, you'r right.  That information was unnecessary at this point.  However, I have tried both Single and Factory in the menuconfig > partition table.  Actually if I choose the option to use the custom partition then I get stuck unable to even get back to menuconfig.  I now get the following traceback
Traceback (most recent call last):
  File "C:/Users/bruce/Desktop/esp-idf/components/partition_table/parttool.py", line 98, in __init__
    partition_table = gen.PartitionTable.from_binary(f.read())
  File "C:\Users\bruce\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 231, in from_binary
    result.append(PartitionDefinition.from_binary(data))
  File "C:\Users\bruce\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 383, in from_binary
    raise InputError("Invalid magic bytes (%r) for partition definition" % magic)
gen_esp32part.InputError: Invalid magic bytes (b'# ') for partition definition

eponymic, 2020-04-20:
At that point, the only thing that works for me is deleting the esp-who directory and re-cloning a fresh copy and rebuilding.  Then I can again get into menuconfig.