espressif/esp-who #147: ESP-IDF:  v4.0.1, esp-who examples fail with "Failed to resolve component  'esp32-camera' "

dalemcbeath, 2020-06-04:
## Environment

- Development Kit:        [ESP-EYE]
- Kit version: []
- Module or chip used:  [ESP-EYE]
- IDF version:                  ESP-IDF:  v4.0.1  (according to hello_world project, running locally)
- Build System:              [idf.py]
- Compiler version        8.2.0
- Operating System:     [macOS]
- (Windows only) environment type: [].
- Using an IDE?:            [No]
- Power Supply:            [USB]

## Problem Description
followed **Quick Start with ESP-WHO**  all the way through the hello_world project, which builds fine and works on the ESP-EYE board via Arduino Serial Monitor.  

repeated process with each of the other included esp_who examples, each gives same error when trying to run   **idf.py menuconfig** or **idf.py -p  /dev/cu.usbserial-1430 flash**.   All tries result in the same error: 

`-- Building ESP-IDF components for target esp32
CMake Error at /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'esp32-camera'.`

I have tried all the examples, only hello_world works.   I did get around a partition size problem by changing the table start from 8000 to 9000.   I have gone through the issue lists and seen a few references to the **"Failed to resolve component"** issue, but none of the suggestions worked, or seemed appropriate, for my situation.  

I have re-downloaded today's (2020-06-04) version of esp-who and tried new copies of those included examples with the same error.  I do note there is an updated component "esp32-camera @ af93185" but I still get the same errors.

thank you, both for any help and for supporting these amazing boards.  This is my first ever issue submission, so I apologize for any errors.  

### Expected Behavior
that all of the stable version examples would work.
### Actual Behavior
only hello_world works.
### Steps to reproduce
1.  follow Quick start on Mac OS X  Catalina (v 10.15.5)
2. build and test hello_world by using terminal: 
    idf.py menuconfig (set SPIRAM to 4)
    idf.py -p  /dev/cu.usbserial-1430 flash
    (worked for me)
3. follow same procedure to copy and build any of the other examples, such as **detection_with_command_line** .  

### Code to reproduce this issue

``` 
>> cd detection_with_command_line
>> idf.py menuconfig

```

```    
                :
                :
(msgs above this look fine, can edit to include if needed)
                : 
                : 
-- Building ESP-IDF components for target esp32
CMake Error at /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'esp32-camera'.
Call Stack (most recent call first):
  /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  /Users/CritterLab/esp/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
```

## Debug Logs

```
cat /Users/CritterLab/ESP32-EYE/detection_with_command_line/build/CMakeFiles/CMakeOutput.log
[... 317 lines cut ...]
    arg [-plugin-opt=/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper] ==> ignore
    arg [-plugin-opt=-fresolution=/var/folders/5p/4l0zgkyd2zv3lf7nfj2xq41r0000gq/T//ccXpqAG7.res] ==> ignore
    arg [-o] ==> ignore
    arg [cmTC_91f9d] ==> ignore
    arg [-L/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0] ==> dir [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0]
    arg [-L/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc] ==> dir [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc]
    arg [-L/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib] ==> dir [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib]
    arg [CMakeFiles/cmTC_91f9d.dir/CMakeCXXCompilerABI.cpp.obj] ==> ignore
  collapse library dir [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0] ==> [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.2.0]
  collapse library dir [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc] ==> [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/lib/gcc]
  collapse library dir [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib] ==> [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/lib]
  implicit libs: []
  implicit dirs: [/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.2.0;/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/lib/gcc;/Users/CritterLab/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/lib]
  implicit fwks: []

```

## Other items if possible

- [x ] sdkconfig file 

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4732774/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

holubcsongor, 2020-06-06:
Please check, if the PATH of the required components is in the sme location as described in CMakeLists.txt.

dalemcbeath, 2020-06-06:
Aloha, Holub, thanks for your help!

Here's the default CmakeLists.txt

`cmake_minimum_required(VERSION 3.5)`
 `  ` 
`set(EXTRA_COMPONENT_DIRS ../../../components)`
`  `
`include($ENV{IDF_PATH}/tools/cmake/project.cmake)`
`project(recognition_solution)`

And IDF_PATH is 

`>> echo $IDF_PATH`
`/Users/CritterLab/esp/esp-idf`

and that directory contains:

`>> ls $IDF_PATH/tools/cmake/`  
           
`build.cmake.                      git_submodules.cmake.                 project.cmake`   
`targets.cmake                   version.cmake                                 component.cmake`
` idf.cmake                          project_description.json.in.            third_party`
`convert_to_cmake.py.       kconfig.cmake.                                run_cmake_lint.sh`
`toolchain-esp32.cmake   crosstool_version_check.cmake      ldgen.cmake`

holubcsongor, 2020-06-15:
set(EXTRA_COMPONENT_DIRS ../../../components)

I said, check this location, I think, that this directory does not exist (more or less ../).
If it would exist, it would appear in components.

dalemcbeath, 2020-06-16:
Yes, that directory did not exist.  That directory on my machine is /Users/

I found a components directory one level back that did contain esp32-camera.

I changed the project CMakeLists.txt file to `set(EXTRA_COMPONENT_DIRS ../components)` then  ran `idf.py menuconfig`.  Same error.  

I changed it back to `set(EXTRA_COMPONENT_DIRS ../../../components)` and created a directory in that location with the component, and tried again.  

`dm > cd /Users/CritterLab/ESP32-EYE/detection_with_command_line`
`dm > cat CMakeLists.txt`

`  cmake_minimum_required(VERSION 3.5)`
`  set(EXTRA_COMPONENT_DIRS ../../../components)`
`  include($ENV{IDF_PATH}/tools/cmake/project.cmake)`
`  project(detection_with_command_line)`

`dm > ls ../../../components`

` esp32-camera`
`dm >`

then ran `idf.py menuconfig`.  Same error.  

`-- Building ESP-IDF components for target esp32
CMake Error at /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'esp32-camera'.
Call Stack (most recent call first):
  /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  /Users/CritterLab/esp/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  /Users/CritterLab/esp/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:7 (project)`

thank you once again for your help, it is appreciated.

mzmm70000, 2020-11-24:
I am assuming you did not clone the `esp-who` repository using the `--recursive` option. The components your build fails to fail are subrepositories and thus their directories will be empty if you don't pull them in via cloning recursively. Could you try again using 

`git clone --recursive https://github.com/espressif/esp-who.git`

...and let us know how that went?

ZumelzuR, 2021-04-21:
same issue here, I tried with 

`git clone --recursive https://github.com/espressif/esp-who.git `

and didn't work.
`
-- Check for working CXX compiler: /Users/czu/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 40ec4530
-- Building ESP-IDF components for target esp32
CMake Error at /Users/czu/esp-who/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'esp32-camera'.`

`

ZumelzuR, 2021-04-21:
Finally I make it work, the esp-info reference is not the same that master branch esp-info at 
https://github.com/espressif/esp-idf

So I have to clone that repo and replace the folder esp-who/esp-inf, then I also I have to check the EXTRA_COMPONENT_DIRS path on CMakeList.txt that is in my project to reference the components folder inside esp-who. 

After that I can make it work.

leoheck, 2022-07-14:
Good job @ZumelzuR, I could not reproduce your steps. 
Do you have a better idea what I could do?

sirajissani, 2022-11-29:
@leoheck I had this same problem as the OP until I downgraded the IDF version from `v5.0-rc2` to `v4.4.3`. See this [note] where the _mDNS_ package has been moved out of the ESP-IDF repo since v5.0.

[note]: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mdns.html

The other solution might be to explicitly add the _mDNS_ as a component (git submodule) to this repo and maintain a fork.

leoheck, 2022-11-29:
Oh, good, I will give it a try again with your tips to check if I can get it working. Thanks @sirajissani

integer-overflown, 2023-06-04:
This thread looks old, but in case someone's still stuck with this issue, here's how I managed to resolve it:
1. Download esp32 component manually from [here](https://github.com/espressif/esp32-camera/releases), I picked the latest release
2. Unpack it to any temporary location and copy the folder into `components` directory of your ESP IDF installation. Note that the name needs to be exactly `esp32-camera`, so make sure to rename the folder before/after copying (i.e. remove the trailing version)
3. I also had to do the same with esp-dl dependency, it can be found here `https://github.com/espressif/esp-dl/tags`
4. Remove your build directory and re-run the `idf.py` command

After these steps I managed to successfully run `idf.py set-target esp32` on my side
I tested it on `v4.4.3`, as I tried @sirajissani suggestion first, but I assume that'll work on later version as well

P.S. IDK why those aren't downloaded automatically, maybe it's a macOS-specific issue

leoheck, 2023-06-04:
thanks I have to check your solution since I could not make it work.

leoheck, 2023-06-12:
I am new to this. But yay, I could make it work this time, with your help on `Ubuntu 23.04`

Here is my fully working setup...
I did not have to make any custom changes, tho.

Remove previously installed stuff
```bash
rm -rf ~/.espressif/
```

First of all, install `esp-idf (4.4.5)`
```bash
git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v4.4.5
./install.sh esp32
. ./export.sh
```

Then this repo (`esp-who`)
```bash
git clone --recursive https://github.com/espressif/esp-who.git
cd esp-who
idf.py set-target esp32
# select the example intended
cd ./examples/motion_detection/terminal
idf.py build
# Plug the device, and check the device.. in my case, it was /dev/ttyUSB0
idf.py -p /dev/ttyUSB0 flash 
idf.py -p /dev/ttyUSB0 monitor
```

[image]