espressif/esp-who #148: Detected camera not supported

mzimmers, 2020-06-17:
Hi -

Trying to use this code on the ESP-EYE...builds and flashes fine (*) but when it runs, I get this:
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5424
ho 0 tail 12 room 4
load:0x40078000,len:12452
load:0x40080400,len:6716
entry 0x400806d4
I (168) psram: This chip is ESP32-D0WD
I (168) spiram: Found 64MBit SPI RAM device
I (168) spiram: SPI RAM mode: flash 80m sram 40m
I (171) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (178) cpu_start: Pro cpu up.
I (182) cpu_start: Application information:
I (187) cpu_start: Project name:     detection_with_command_line
I (194) cpu_start: App version:      v0.9.3-46-g2470e47
I (199) cpu_start: Compile time:     Jun 17 2020 12:06:43
I (206) cpu_start: ELF file SHA256:  582162204e0d2984...
I (212) cpu_start: ESP-IDF:          v4.0.1-dirty
I (217) cpu_start: Starting app cpu, entry point is 0x400813f8
0x400813f8: call_start_cpu1 at C:/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (1107) spiram: SPI SRAM memory test OK
I (1108) heap_init: Initializing. RAM available for dynamic allocation:
I (1108) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1114) heap_init: At 3FFB48B0 len 0002B750 (173 KiB): DRAM
I (1120) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1127) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1133) heap_init: At 40090B30 len 0000F4D0 (61 KiB): IRAM
I (1140) cpu_start: Pro cpu start user code
I (1145) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1165) spi_flash: detected chip: generic
I (1165) spi_flash: flash io: qio
I (1165) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1174) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1184) sccb: pin_sda 26 pin_scl 27

E (130204) camera: Detected camera not supported.
E (130204) camera: Camera probe failed with error 0x20004
E (130204) app_camera: Camera init failed with error 0x20004
E (130204) app_process: Camera capture failed
E (130214) app_process: Camera capture failed
E (130214) app_process: Camera capture failed`

The last line goes on forever. Any ideas what's wrong?

(*) the flash is fine, as long as I do just an app-flash. If I attempt a "normal" flash, I get an error:

> esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: partition_table/partition-table.bin

Am I wrong to expect this to work out of the box?

Thanks...

TiramisuJ, 2020-06-18:
You can try to enter the “bootloader config” in “menuconfig” and then set the “bootloader log verbosity” to “warning”.

mzimmers, 2020-06-18:
Thank you for the suggestion, but that did not solve the problem.

mzimmers, 2020-06-18:
According to an old document I found on adafruit, I need to change my IDF_PATH variable to the idf under esp-who. Is this true? I don't see anything about it in the readme here.

TiramisuJ, 2020-06-19:
Yes, you need to make sure that the IDF version under IDF_PATH is the same as the IDF version under esp-who.

mzimmers, 2020-06-19:
OK...same error. Did a fullclean first, then a build, then flash.

TiramisuJ, 2020-06-19:
If you are running the “detection_with_command_line” example, have you set the “camera pin” option in “menuconfig” to “ESP_EYE DevKit” ?

mzimmers, 2020-06-19:
Aha...that was it. Thanks, TiramisuJ.

Seems to me that this little detail belongs in one of the project readme.md files...

closed 2020-06-19 by mzimmers

TiramisuJ, 2020-06-20:
Thank you for your suggestion, we will update the documents in time