espressif/esp-who #151: OV7725: E (3410) camera: Detectet camera not Supported.

dblatt87, 2020-08-03:
Hello

I try to get an OV7725 running on an ESP32-CAM from ai thinker.
To check the cameras I just load the "camera_web_server" example. 
OV2640 and OV5640 are working well, OV7725 not.
I read here #60 it should work with all my cameras and the Board.

OV7725 is selectet in camera settings, also OV2640 and OV5640.
These are the errors:

E (3410) camera: Detectet camera not Supported.
E (3410) camera: Camera probe failed with error 0x20001
E (3410) app_camera: Camera init failed with error 0x20001

I also tried it with software I2C and selecting only OV7725, no luck.

Does someone know from similar errors?
Are there cameras with different pin assignments?

Thanks in advance

henkiejan1, 2020-10-01:
Hi,

May i aks you how you got the OV5640 working? Just only plugged in and select OV5640? The camera module is detected but does not got any image. The camera is detected and can select a lot more properties and resolutions but i got the message that the there is something not arrived on time..

> I try to get an OV7725 running on an ESP32-CAM from ai thinker.
> To check the cameras I just load the "camera_web_server" example.
> OV2640 and OV5640 are working well, OV7725 not.

Could you give me the steps please?

Thank you in advanced!

henkiejan1, 2020-11-01:
> I try to get an OV7725 running on an ESP32-CAM from ai thinker.
> To check the cameras I just load the "camera_web_server" example.
> OV2640 and OV5640 are working well, OV7725 not.

Could you please help me to got the OV5640 get working? Thank you very much!

dblatt87, 2021-08-28:
@yehangyang, thanks for asking. 
Yes it worked with [](https://github.com/espressif/esp32-camera) and a right ov7725 (seems I got one with an other pin asignment...)

@henkiejan1, It was a while ago... I just uploaded an esp-who example, I think back then I was still working with Arduino

closed 2021-08-28 by dblatt87