espressif/esp-who #156: 移植到ESP32S2芯片

ZhongShangWen, 2020-09-12:
在移植后设备在初始化摄像头时芯片会一直重启
[image]

我用的硬件时ESP32-S2-Kaluga-1 V1.2开发板 和配套的 VO2640摄像头模块

通过追踪发现代码最终死在 SCCB写配置的地方
[image]
[image]
[image]

不知道什么原因导致的重启

ChenZhengwei20190741, 2020-09-14:
kaluga 上 camera 硬件 reset 的管脚未连接到 s2 上，会不会你配置reset管脚上出错导致的？

ChenZhengwei20190741, 2020-09-14:
[image]
1.2板子上连接摄像头上的45，46管脚是 strapping pin ，会对下载和启动阶段有影响，所以你可以尝试更改管脚试试，图中红框内的两个电阻去掉。然后将摄像头上标注的D2,D3管脚连到其他管脚（如36 37）试试。

ZhongShangWen, 2020-09-15:
@ChenZhengwei20190741 Hi，上午好，这个是我的引脚配置
[image]

ChenZhengwei20190741, 2020-09-15:
你提个分支吧，我这边 pull 下来试试

ZhongShangWen, 2020-09-15:
好的

ZhongShangWen, 2020-09-15:
@ChenZhengwei20190741  Hi,下午好，我将代码推到了gitee 地址如下
https://gitee.com/shangWen-gitee/esp32_-s2_camera.git
工程目录下README.md有说明

ChenZhengwei20190741, 2020-09-15:
刚好我这边最近也在适配，我这边会尽快帮你解决。

ChenZhengwei20190741, 2020-09-16:
hi，你 JTAG 功能需要使用到吗，目前是这样的，JTAG 引脚可能导致这关于解决方式是
···
espefuse.py burn_efuse HARD_DIS_JTAG
···
注意，这个是不可逆的。这个会永久不能使用 JTAG ,不会影响其他功能。

但我解决完发现，你的 camera 适配还是有点问题。

ZhongShangWen, 2020-09-17:
JTAG这个功能可以不使用
因为之前一直卡在了重启的问题上，所以相机初始化后面的功能都还没验证，请问你有发现什么异常的问题吗？

ZhongShangWen, 2020-09-18:
@ChenZhengwei20190741 Hi,下午好，摄像头重启的问题已经通过你给的方法解决了，现在有一个新的问题，libdl.a
这个库你们之前是拿esp32的工具链编译的，所以我在用ESP32S2编译时是用不了的，能麻烦生成一份ESP32S2工具链生成的库吗？
[image]

ZhongShangWen, 2020-09-18:
[image]
这文件夹下的.a文件都需要用esp32s2的工具链编译生成

closed 2022-09-21 by Auroragan