espressif/esp-who #164: Abort was called at PC 0x40084dfe on core 0

jhancharler, 2020-11-29:
I've tried flashing the camera_web_server example to my ESP32-CAM.

It seems to flash OK but the monitor output is very strange and I don't think it's working correctly.

I'm using the latest ESP-IDF. I connect GPIO0 to GND when flashing, and take it off for when running the application. I selected ESP32 AI-THINKER board on menuconfig.

I seem to be seeing this issue with other source codes too so there must be something I'm missing. 

Hope someone can help!

Here's my monitor output:

```
ELF file SHA256: ee2c7aadea4b31dc

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:5932
load:0x40078000,len:14988
ho 0 tail 12 room 4
load:0x40080400,len:4028
0x40080400: _init at ??:?

entry 0x40080648
I (931) psram: This chip is ESP32-D0WD
I (932) spiram: Found 64MBit SPI RAM device
I (932) spiram: SPI RAM mode: flash 80m sram 40m
I (935) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (942) cpu_start: Pro cpu up.
I (946) cpu_start: Starting app cpu, entry point is 0x4008156c
0x4008156c: call_start_cpu1 at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/esp_system/port/cpu_start.c:126

I (937) cpu_start: App cpu up.
I (1833) spiram: SPI SRAM memory test OK
I (1840) cpu_start: Pro cpu start user code
I (1840) cpu_start: cpu freq: 240000000
I (1840) cpu_start: Application information:
I (1843) cpu_start: Project name:     camera_web_server
I (1849) cpu_start: App version:      v0.9.3-60-g73d4e2d
I (1855) cpu_start: Compile time:     Nov 29 2020 14:46:53
I (1862) cpu_start: ELF file SHA256:  ee2c7aadea4b31dc...
I (1868) cpu_start: ESP-IDF:          v4.3-dev-1901-g178b122c1-dirty
I (1875) heap_init: Initializing. RAM available for dynamic allocation:
I (1882) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1888) heap_init: At 3FFBD450 len 00022BB0 (138 KiB): DRAM
I (1894) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1901) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1907) heap_init: At 4009B808 len 000047F8 (17 KiB): IRAM
I (1914) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1922) spi_flash: detected chip: generic
I (1926) spi_flash: flash io: qio
I (1931) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (2024) wifi:wifi driver task: 3ffc5fd0, prio:23, stack:6656, core=0
I (2024) system_api: Base MAC address is not set
I (2024) system_api: read default base MAC address from EFUSE
I (2044) wifi:wifi firmware version: b9e8721
I (2044) wifi:wifi certification version: v7.0
I (2044) wifi:config NVS flash: enabled
I (2044) wifi:config nano formating: disabled
I (2044) wifi:Init data frame dynamic rx buffer num: 32
I (2054) wifi:Init management frame dynamic rx buffer num: 32
I (2054) wifi:Init management short buffer num: 32
I (2064) wifi:Init static tx buffer num: 16
I (2064) wifi:Init tx cache buffer num: 32
I (2064) wifi:Init static rx buffer size: 1600
I (2074) wifi:Init static rx buffer num: 10
I (2074) wifi:Init dynamic rx buffer num: 32
I (2084) wifi_init: rx ba win: 6
I (2084) wifi_init: tcpip mbox: 32
I (2084) wifi_init: udp mbox: 6
I (2094) wifi_init: tcp mbox: 6
I (2094) wifi_init: tcp tx win: 5744
I (2094) wifi_init: tcp rx win: 5744
I (2104) wifi_init: tcp mss: 1440
I (2104) wifi_init: WiFi IRAM OP enabled
I (2114) wifi_init: WiFi RX IRAM OP enabled
I (2124) camera wifi: wifi_init_softap finished.SSID:ESP32-Camera password:
I (2124) camera wifi: wifi_init_sta finished.
I (2134) camera wifi: connect to ap SSID:VM2174627 password:Goldenlight71
I (2214) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (2214) wifi:mode : sta (c4:4f:33:3a:5d:a5) + softAP (c4:4f:33:3a:5d:a6)
I (2224) wifi:enable tsf
I (2224) wifi:Total power save buffer number: 8
I (2224) wifi:Init max length of beacon: 752/752
I (2234) wifi:Init max length of beacon: 752/752
I (2234) wifi:Set ps type: 0

I (2244) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2244) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2254) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2264) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2274) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2284) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2294) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2304) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2314) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2324) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2334) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2344) sccb: pin_sda 26 pin_scl 27

abort() was called at PC 0x40084dfe on core 0
0x40084dfe: lock_acquire_generic at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/newlib/locks.c:140

Backtrace:0x4008ddae:0x3ffbf4b0 0x4008e5dd:0x3ffbf4d0 0x40095a46:0x3ffbf4f0 0x40084dfe:0x3ffbf560 0x40084f5d:0x3ffbf590 0x4016466d:0x3ffbf5b0 0x40167a63:0x3ffbf8c0 0x40171cf3:0x3ffbf8f0 0x400943bb:0x3ffbf920 0x4011893d:0x3ffbf970 0x400def12:0x3ffbf9a0 0x400de3fb:0x3ffbf9f0 0x400decb5:0x3ffbfa30 0x400db619:0x3ffbfa60 0x400db262:0x3ffbfae0 0x40171803:0x3ffbfb00 0x40090ee1:0x3ffbfb20
0x4008ddae: panic_abort at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/esp_system/panic.c:367

0x4008e5dd: esp_system_abort at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/esp_system/system_api.c:106

0x40095a46: abort at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/newlib/abort.c:46

0x40084dfe: lock_acquire_generic at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/newlib/locks.c:140

0x40084f5d: _lock_acquire_recursive at C:/DEV/git-sdk/git-sdk-64/esp/esp-idf/components/newlib/locks.c:168

0x4016466d: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/l
```

me-no-dev, 2020-11-30:
unfortunately the backtrace does not give enough information on how did it get to that abort. Did you try with other IDF versions?

jhancharler, 2020-11-30:
@me-no-dev I was able to get this to work on a Linux machine. Still unsure why this issue occurs on Windows. 

I haven't tried other IDF_Versions but I will stay with Linux for now

me-no-dev, 2020-11-30:
is here any way that you can post the elfs from both machines? as long as they are the same code, there should not be a difference, but since you find such, it could help us find what is wrong :)

jhancharler, 2020-12-01:
Do you mean the camera_web_server.elf files?

I can post my Windows .elf file but I am unsure how I can do that with my Linux machine as I only have access to it via the terminal.

Windows (non-working):
[camera_web_server.zip](https://github.com/espressif/esp-who/files/5623105/camera_web_server.zip)

me-no-dev, 2020-12-01:
I guess you are using SSH (PuTTY). You can also use WinSCP to access the file system on the server through SSH :) We need both files so we can diff them against each other.

jhancharler, 2020-12-01:
Yes I was, thanks for the tip regarding WinSCP :).

Here is the .elf file from the Linux machine:

[camera_web_server_linux(working).zip](https://github.com/espressif/esp-who/files/5623346/camera_web_server_linux.working.zip)

me-no-dev, 2020-12-01:
Thanks @jhancharler :)

me-no-dev, 2020-12-02:
@jhancharler could you please confirm that both computers have the exact same IDF and toolchain version? Have you done any modification to the example itself? The error hints at trying to print from within ISR.

jhancharler, 2020-12-08:
Hey @me-no-dev  sorry for delayed response. What's the best way to do this?

me-no-dev, 2020-12-09:
when you `idf.py build` the project, one of the first lines will show IDF version (also while booting) and the toolchain version used.

jhancharler, 2020-12-13:
@me-no-dev When I type `idf.py build` I don't see the IDF version. This is what I see instead (Same for both Windows and my Linux machine):

[image]

Anyway, I found this command `idf.py --version` and that yields the following (on the non-working Windows machine):

ESP VERSION: `ESP-IDF v4.3-dev-1901-g178b122c1-dirty`

With regards to the toolchain, this is the information I think might be helpful:

`Adding ESP-IDF tools to PATH...
    C:\DEV\.espressif1\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin

    C:\DEV\.espressif1\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin

    C:\DEV\.espressif1\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin

    C:\DEV\.espressif1\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin

    C:\DEV\.espressif1\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin

    C:\DEV\.espressif1\tools\cmake\3.16.4\bin

    C:\DEV\.espressif1\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin

    C:\DEV\.espressif1\tools\ninja\1.10.0\

    C:\DEV\.espressif1\tools\idf-exe\1.0.1\

    C:\DEV\.espressif1\tools\ccache\3.7\

    C:\DEV\.espressif1\tools\dfu-util\0.9\dfu-util-0.9-win64

    C:\DEV\.espressif1\python_env\idf4.3_py3.9_env\Scripts

    C:\DEV\git-sdk\git-sdk-64\esp\esp-idf\tools`

Ok, now on the working machine it's this: `ESP-IDF version: ESP-IDF v4.3-dev-1901-g178b122c1-dirty`

Toolchain version: 
[image]

I don't have much time left with this Windows machine and will be moving to another machine soon so I hope this information is enough for you guys to debug the issue.

Thanks for all of your work, really enjoying the esp32 :).

pkjains, 2020-12-14:
I received an ESP_EYE and was trying to run face_detection_with_command_line application. I am seeing the below back trace which looks pretty identical to this issue.

The issue seems to be coming from esp_camera_init()

--------

```
ELF file SHA256: 557bbb76f7c17546

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7388
load:0x40078000,len:15504
load:0x40080400,len:4436
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.3-dev-1901-g178b122c1-dirty 2nd stage bootloader
I (29) boot: compile time 18:18:43
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) qio_mode: Enabling default flash chip QIO
I (46) boot.esp32: SPI Speed      : 80MHz
I (51) boot.esp32: SPI Mode       : QIO
I (55) boot.esp32: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 factory          factory app      00 00 00010000 00200000
I (84) boot:  1 nvs              WiFi data        01 02 00310000 00004000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2dd4c (187724) map
I (174) esp_image: segment 1: paddr=0x0003dd74 vaddr=0x3ffb0000 size=0x022a4 (  8868) load
I (177) esp_image: segment 2: paddr=0x00040020 vaddr=0x400d0020 size=0x246c0 (149184) map
0x400d0020: _stext at ??:?

I (230) esp_image: segment 3: paddr=0x000646e8 vaddr=0x3ffb22a4 size=0x015e8 (  5608) load
I (232) esp_image: segment 4: paddr=0x00065cd8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/praveenkjain/code/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

I (237) esp_image: segment 5: paddr=0x000660e4 vaddr=0x40080404 size=0x0e634 ( 58932) load
I (277) boot: Loaded app from partition at offset 0x10000
I (277) boot: Disabling RNG early entropy source...
I (289) psram: This chip is ESP32-D0WD
I (289) spiram: Found 64MBit SPI RAM device
I (289) spiram: SPI RAM mode: flash 80m sram 80m
I (292) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (299) cpu_start: Pro cpu up.
I (303) cpu_start: Starting app cpu, entry point is 0x40081268
0x40081268: call_start_cpu1 at /Users/praveenkjain/code/esp/esp-idf/components/esp_system/port/cpu_start.c:126

I (295) cpu_start: App cpu up.
I (806) spiram: SPI SRAM memory test OK
I (814) cpu_start: Pro cpu start user code
I (814) cpu_start: cpu freq: 240000000
I (814) cpu_start: Application information:
I (817) cpu_start: Project name:     face_detection_with_command_lin
I (824) cpu_start: App version:      v0.9.3-60-g73d4e2d
I (829) cpu_start: Compile time:     Dec 14 2020 18:18:38
I (836) cpu_start: ELF file SHA256:  557bbb76f7c17546...
I (842) cpu_start: ESP-IDF:          v4.3-dev-1901-g178b122c1-dirty
I (849) heap_init: Initializing. RAM available for dynamic allocation:
I (856) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[... 37 lines cut ...]

0x400ea4a5: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x400ed87b: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x400f467b: esp_log_writev at /Users/praveenkjain/code/esp/esp-idf/components/log/log.c:189

0x4008dad3: esp_log_write at /Users/praveenkjain/code/esp/esp-idf/components/log/log.c:199

0x400de979: i2c_param_config at /Users/praveenkjain/code/esp/esp-idf/components/driver/i2c.c:667

0x400d7220: SCCB_Init at /Users/praveenkjain/code/esp/esp-who/components/esp32-camera/driver/sccb.c:52

0x400d6163: camera_probe at /Users/praveenkjain/code/esp/esp-who/components/esp32-camera/driver/camera.c:977
 (inlined by) camera_probe at /Users/praveenkjain/code/esp/esp-who/components/esp32-camera/driver/camera.c:959

0x400d6f49: esp_camera_init at /Users/praveenkjain/code/esp/esp-who/components/esp32-camera/driver/camera.c:1377

0x400d60a5: app_camera_init() at /Users/praveenkjain/code/esp/esp-who/examples/single_chip/face_detection_with_command_line/build/../main/app_camera.cpp:69

0x400d5d9f: app_main at /Users/praveenkjain/code/esp/esp-who/examples/single_chip/face_detection_with_command_line/build/../main/app_main.cpp:38

0x400f443e: main_task at /Users/praveenkjain/code/esp/esp-idf/components/freertos/port/port_common.c:133

```

me-no-dev, 2020-12-29:
can you please pull the latest esp32-camera driver and give it another shot? I just merged a possible fix for this :)

pkjains, 2021-01-03:
I just tested the face_detection_with_command_line with latest code base from just now. It seems to work just fine now.

> can you please pull the latest esp32-camera driver and give it another shot? I just merged a possible fix for this :)

closed 2022-06-24 by Auroragan