espressif/esp-who #176: ESp-WHO Examples Errors

onderbedirhan, 2021-07-15:
Hello, I want to do image processing projects with ESP-EYE microcontroller. I installed the ESP-IDF environment on VS Code with the Espressif IDF plugin. Then I cloned the ESP-WHO library to the directory where the ESP-IDF folder is located, via git. However, when I try to open the face-recognition-solution project in the examples folder in the ESP-WHO library, I get many errors about the libraries not being defined. If there is an extra setting that I need to do or a part that I have made wrong, I would be very happy if you could help me with this.
[image]
[image]
[image]

FalconCode-Ux, 2021-07-27:
From my experience, VS code is an absolute nightmare when working with esp-who. So, I advise you to use ESP-IDF directly from terminal, it's easier in my opinion and clearer . Also the commands are straightforward: "idf.py build" to build the project, "ïdf.py fullclean" to remove all the build files, "idf.py -p [port] flash" to program your board and "idf.py -p [port] monitor" to use serial monitor.

ESP-YHY, 2022-03-03:
Hi @onderbedirhan, Please check out the latest master.

closed 2022-09-21 by Auroragan