espressif/esp-who #184: ESP-WHO example for ESP-EYE errors

waymousa, 2021-11-02:
After building the "human_face_recognition\terminal" I flash and upload the code but bet the following message:

E (1846) gpio: gpio_install_isr_service(460): GPIO isr service already installed

A quick press on the "boot" button generates the following output in the terminal:

W (172686) cam_hal: Failed to get the frame on time!

After which there is a panic and it reboots.  

I'm not a c programmer by trade but I think the GPIO failure is sopping the button presses from being detected.  

Can anyone help with this issue?

TiramisuJ, 2021-11-24:
@waymousa  You can fix this problem temporarily by defining “FB_GET_TIMEOUT”(in esp-who/components/esp32-camera/driver/esp_camera.c  line297)  to "portMAX_DELAY"

ESP-YHY, 2022-03-03:
Hi @waymousa, this issue should be solved now. Please try latest on master again.

closed 2022-03-14 by Auroragan