espressif/esp-who #186: Face detection and face recognition not working

Jen0055, 2021-11-08:
Can someone help me i have error while running the esp32 cam module. Face detection and face recognition is not working 
p32 cam module. Face detection and face recognition is not working 

Item psram alloc failed. Size: 126720 x 1
matrix3du item alloc failed.
dl_matrix3du_alloc failed

yehangyang, 2021-11-19:
It seems you still use the old version. Please pull the latest master.

Jen0055, 2021-11-19:
What do you mean sir ? old version of esp32 cam ? I already used the latest
version sir but still same result. Also the PSRAM is not also working. We
have 2 esp32 cam module but they have the same error. Hope you can help me
sir.

Thank you

yehangyang, 2021-11-19:
@Jen0055 , have you update submodule as well? Because "dl_matrix3du_alloc" is in the old version of ESP-DL and ESP-WHO.
`git submodule update --init`

Jen0055, 2021-11-19:
How can I do that sir? update the submodule

yehangyang, 2021-11-19:
Open a Terminal
Enter the ESP-WHO repository
type `git submodule update --init` and click enter key

Jen0055, 2021-11-19:
Thank you for your time sir. I'm still dont know how to do it sir but I
will try

khaoulaMOS, 2021-11-23:
[image]
anyone can help please

yehangyang, 2021-11-24:
@khaoulaMOS, Please try to update the toolchain by `. {IDF_PATH}/install.bat`.

khaoulaMOS, 2021-11-24:
[image]
[image]
@yehangyang  thank you but still the same error and this is new one

yehangyang, 2021-11-24:
@khaoulaMOS, this issue has been proposed [here](https://github.com/espressif/esp-who/issues/184). We are working on it.

TiramisuJ, 2021-11-24:
@khaoulaMOS , You can fix this problem temporarily by defining “FB_GET_TIMEOUT”(in esp-who/components/esp32-camera/driver/esp_camera.c  line297)  to "portMAX_DELAY"

closed 2022-02-28 by Auroragan