espressif/esp-who #187: Face detection not working(esp32s3 resetting continuously )

nikhil-robinson, 2021-11-26:
I have a custom build esp32s3 board and i am trying to run face detection example code. I am able to successfully upload code after configuring the custom camera pins in menu config  but it is showing the following error keeps on repeating  while running the code.

error output 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1858
load:0x403b6000,len:0xb58
load:0x403ba000,len:0x2fc4
entry 0x403b61e4
I (25) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (25) boot: compile time 14:09:48
I (26) boot: chip revision: 0
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
W (54) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 factory          factory app      00 00 00010000 003c0000
I (79) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (86) boot:  2 fr               unknown          20 20 003e0000 00020000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=5fabch (391868) map
I (166) esp_image: segment 1: paddr=0006fae4 vaddr=3fc932a0 size=00534h (  1332) load
I (166) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=79e64h (499300) map
I (247) esp_image: segment 3: paddr=000e9e8c vaddr=3fc937d4 size=036d0h ( 14032) load
I (250) esp_image: segment 4: paddr=000ed564 vaddr=40374000 size=0f298h ( 62104) load
I (265) esp_image: segment 5: paddr=000fc804 vaddr=50000000 size=00010h (    16) load
I (271) boot: Loaded app from partition at offset 0x10000
I (272) boot: Disabling RNG early entropy source...
W (274) bootloader_random: RNG for ESP32-S3 not currently supported
I (281) opi psram: vendor id : 0x00 (UNKNOWN)
I (286) opi psram: dev id    : 0x03 (generation 4)
I (291) opi psram: density   : 0x07 (256 Mbit)
I (296) opi psram: good-die  : 0x01 (Pass)
I (301) opi psram: Latency   : 0x01 (Fixed)
I (306) opi psram: VCC       : 0x01 (3V)
I (311) opi psram: SRF       : 0x00 (Slow Refresh)
I (316) opi psram: BurstType : 0x01 ( Wrap)
I (321) opi psram: BurstLen  : 0x03 (1024 Byte)
I (326) opi psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (332) opi psram: DriveStrength: 0x03 (1/8)
W (337) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (348) MSPI Timing: tuning fail, best point is fallen back to index 1
I (354) spiram: Found 256MBit SPI RAM device
I (359) spiram: SPI RAM mode: sram 80m
I (364) spiram: PSRAM initialized, cache is in normal (1-core) mode. 

As per various troubleshooting steps we tried  changing usb cable , giving power directly etc still the error persist.

i am using esp-idf 4.4  in vscode on windows 10

any help would be appreciated thank you.

TiramisuJ, 2021-11-29:
Hi @nikhil-robinson, 
1. Please make sure the SPIRAM of your board is octal mode or quad mode，then set the corresponding options in “idf.py menuconfig -> Component config -> ESP32S3-Specific -> Mode (QUAD/OCT) of SPI RAM chip in use”.
2. If your configuration is right, Please update your esp-idf to the master branch.

ESP-YHY, 2022-03-07:
Hi @nikhil-robinson, have this problem been solved?

closed 2022-03-14 by Auroragan