espressif/esp-who #188: Camera pin configuration issue (esp32s3)

nikhil-robinson, 2021-11-29:
I am testing the human face detection example program on esp32s3 custom board I have mapped my camera pins as bellow on my custom pcb,
xclk - 48
SIOD - 8
SIOC - 3
VSYNC - 46
HREF - 21
PCLK -39
Y2 -41
Y3 -43
Y4 -44
Y5 -42
Y6 - 40
Y7 -38
Y8-45
Y9 -47

BUT when i try to map this pins in the menu config it is auto resetting in vscode menuconfig . So i tried it with cmd and it prompts that i can only use pins from 0-39 / 0-33. 
[image]

My question is that 
1. is it  only possible to assign pins from 0-39 for camera configuration or is it a compatibility issue
2. should i fully remap my custom pcb with 0-39 pin configuration or is there a workaround   

i am using idf version 4.4 on windows 10 

Any help would be appreciated 
Thank you

TiramisuJ, 2021-11-29:
Hi @nikhil-robinson
1. This pin limit is for ESP32. We have not modified this limit for ESP32-S3. You can delete the range limit for each pins in "esp-who/components/modules/Kconfig, from line92, line100, ... to line212"
2. You don't need to remap your custom pcb. you can refer to the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) for more pin information

nikhil-robinson, 2021-11-29:
Hi @TiramisuJ 

Thanks for the replay i will give this a try.

nikhil-robinson, 2021-11-29:
@TiramisuJ  i tried the above said solution but now the program is not building showing this error 

[image]

i have not changed anything else just edited the kconfig file  i am able to successfully to map my pins in the menuconfig but when i build the program this error is showing up

TiramisuJ, 2021-11-29:
@nikhil-robinson This is caused by a bug in esp-sr. you can close the wakenet (idf.py menuconfig -> ESP Speech Recognition -> use wakenet) and multinet (idf.py menuconfig -> ESP Speech Recognition -> use multinet), and use "idf.py clean" after closing. then build again.

nikhil-robinson, 2021-11-29:
@TiramisuJ  That worked i am able to build the program . But when i run the code the bellow error is showing.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1858
load:0x403b6000,len:0xb58
load:0x403ba000,len:0x2fc4
entry 0x403b61e4
I (25) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (25) boot: compile time 19:44:19
I (25) boot: chip revision: 0
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 factory          factory app      00 00 00010000 003c0000
I (79) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (86) boot:  2 fr               unknown          20 20 003e0000 00020000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=5facch (391884) map
I (166) esp_image: segment 1: paddr=0006faf4 vaddr=3fc92880 size=00524h (  1316) load
I (166) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=79e14h (499220) map
I (247) esp_image: segment 3: paddr=000e9e3c vaddr=3fc92da4 size=03418h ( 13336) load
I (250) esp_image: segment 4: paddr=000ed25c vaddr=40374000 size=0e874h ( 59508) load
I (264) esp_image: segment 5: paddr=000fbad8 vaddr=50000000 size=00010h (    16) load
I (270) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
W (273) bootloader_random: RNG for ESP32-S3 not currently supported
I (280) spiram: Found 16MBit SPI RAM device
I (285) spiram: SPI RAM mode: sram 80m
I (289) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (296) cpu_start: Pro cpu up.
I (300) cpu_start: Starting app cpu, entry point is 0x40375364
0x40375364: call_start_cpu1 at C:/Users/acer/Desktop/esp-idf/components/esp_system/port/cpu_start.c:156

TiramisuJ, 2021-11-30:
This looks like a bug in esp-idf a long time ago, you can update your esp-idf to the master branch and rebuild your project.

nikhil-robinson, 2021-11-30:
@TiramisuJ But i am using the latest idf 4.4

TiramisuJ, 2021-11-30:
@nikhil-robinson  Does the latest idf master branch have the same problem ?

nikhil-robinson, 2021-11-30:
@TiramisuJ  yes this error is showing on the latest master branch

TiramisuJ, 2021-11-30:
Is it useful to modify the SPIRAM mode from octal mode to quad mode ?

nikhil-robinson, 2021-11-30:
@TiramisuJ  SPIRAM is set on quad mode but this doesn't solve the problem

TiramisuJ, 2021-11-30:
Could you post your complete error logs and the your hardware information(chip version, spiram, flash... etc)?

nikhil-robinson, 2021-12-06:
@TiramisuJ  Does this error has any relation with ram . like is there any ram specification for esp-who

nikhil-robinson, 2021-12-06:
Hardware Specs 
ESP32s3 N8R2
camera OV2 320x240
psram 2 quad spi
8 mb flash

error log:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1858
load:0x403b6000,len:0xb58
load:0x403ba000,len:0x2fc4
entry 0x403b61e4
I (25) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (25) boot: compile time 19:44:19
I (25) boot: chip revision: 0
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (64) boot: ## Label Usage Type ST Offset Length
I (71) boot: 0 factory factory app 00 00 00010000 003c0000
I (79) boot: 1 nvs WiFi data 01 02 003d0000 00004000
I (86) boot: 2 fr unknown 20 20 003e0000 00020000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=5facch (391884) map
I (166) esp_image: segment 1: paddr=0006faf4 vaddr=3fc92880 size=00524h ( 1316) load
I (166) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=79e14h (499220) map
I (247) esp_image: segment 3: paddr=000e9e3c vaddr=3fc92da4 size=03418h ( 13336) load
I (250) esp_image: segment 4: paddr=000ed25c vaddr=40374000 size=0e874h ( 59508) load
I (264) esp_image: segment 5: paddr=000fbad8 vaddr=50000000 size=00010h ( 16) load
I (270) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
W (273) bootloader_random: RNG for ESP32-S3 not currently supported
I (280) spiram: Found 16MBit SPI RAM device
I (285) spiram: SPI RAM mode: sram 80m
I (289) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (296) cpu_start: Pro cpu up.
I (300) cpu_start: Starting app cpu, entry point is 0x40375364
0x40375364: call_start_cpu1 at C:/Users/acer/Desktop/esp-idf/components/esp_system/port/cpu_start.c:156

TiramisuJ, 2021-12-07:
@nikhil-robinson Does this error still appear if you close the spiram in menuconfig ?

nikhil-robinson, 2021-12-07:
@TiramisuJ  disabling the spiram still gives error 

error log 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1858
load:0x403b6000,len:0xb58
load:0x403ba000,len:0x2fc4
entry 0x403b61e4
I (24) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (25) boot: compile time 10:24:47
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 factory          factory app      00 00 00010000 003c0000
I (78) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (86) boot:  2 fr               unknown          20 20 003e0000 00020000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=5f6bch (390844) map
I (165) esp_image: segment 1: paddr=0006f6e4 vaddr=3fc922c0 size=00934h (  2356) load
I (166) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=79644h (497220) map
I (246) esp_image: segment 3: paddr=000e966c vaddr=3fc92bf4 size=02f28h ( 12072) load
I (248) esp_image: segment 4: paddr=000ec59c vaddr=40374000 size=0e2b8h ( 58040) load
I (263) esp_image: segment 5: paddr=000fa85c vaddr=50000000 size=00010h (    16) load
I (269) boot: Loaded app from partition at offset 0x10000
I (269) boot: Disabling RNG early entropy source...
W (272) bootloader_random: RNG for ESP32-S3 not currently supported
I (279) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x403752d0
0x403752d0: call_start_cpu1 at C:/Users/acer/Desktop/esp-idf/components/esp_system/port/cpu_start.c:156

TiramisuJ, 2021-12-07:
Can you run `$IDF_PATH/examples/get-started/hello_world` this example successfully ?

nikhil-robinson, 2021-12-07:
@TiramisuJ  yes i can run the hello world program and it works fine

TiramisuJ, 2021-12-07:
This error should have occurred in the boot phase, but I can't see the specific error print from your log. When this error occurs, the code of esp-who should not have been run, so I suspect that the configuration of menuconfig caused this problem. If you replace the sdkconfig file in hello world with the sdkconfig in your project of esp-who, can hello world still run successfully.

nikhil-robinson, 2021-12-07:
@TiramisuJ  I tried it but bellow error is showing

FAILED: partition_table/partition-table.bin 
cmd.exe /C "cd /D C:\Users\acer\Desktop\esp-who\examples\human_face_detection\terminal\build\esp-idf\partition_table && C:\Users\acer\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/acer/Desktop/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB C:/Users/acer/Desktop/esp-who/examples/human_face_detection/terminal/partitions.csv C:/Users/acer/Desktop/esp-who/examples/human_face_detection/terminal/build/partition_table/partition-table.bin && C:\Users\acer\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E echo "Partition table binary generated. Contents:" && C:\Users\acer\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E echo ******************************************************************************* && C:\Users\acer\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/acer/Desktop/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB C:/Users/acer/Desktop/esp-who/examples/human_face_detection/terminal/build/partition_table/partition-table.bin && C:\Users\acer\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E echo *******************************************************************************"
Partitions defined in 'C:/Users/acer/Desktop/esp-who/examples/human_face_detection/terminal/partitions.csv' occupy 4.0MB of flash (4194304 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.
[4/1038] Running move model...
'rm' is not recognized as an internal or external command,
operable program or batch file.
'cp' is not recognized as an internal or external command,
operable program or batch file.
'cp' is not recognized as an internal or external command,
operable program or batch file.
C:/Users/acer/Desktop/esp-who/examples/human_face_detection/terminal/sdkconfig
C:/Users/acer/Desktop/esp-who/components/esp-sr/model
C:/Users/acer/Desktop/esp-who/components/esp-sr/model/wakenet_model/alexa8
C:/Users/acer/Desktop/esp-who/components/esp-sr/model/multinet_model/mn3cn
Recommended model partition size:  500KB
[5/1038] cmd.exe /C "cd /D C:\Users\acer\Desktop\esp-who\examples\human_face_d...bin/cmake.exe -P C:/Users/acer/Desktop/esp-idf/tools/cmake/scripts/fail.cmake"FAILED: esp-idf/esp-sr/CMakeFiles/spiffs_model_bin
cmd.exe /C "cd /D C:\Users\acer\Desktop\esp-who\examples\human_face_detection\terminal\build\esp-idf\esp-sr && C:\Users\acer\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E echo "Failed to create SPIFFS image for partition 'model'. " "Check project configuration if using the correct partition table file." && C:\Users\acer\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E remove C:/Users/acer/Desktop/esp-who/examples/human_face_detection/terminal/build/esp-idf/esp-sr/nSyic.cmake && C:\Users\acer\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" C:/Users/acer/.espressif/tools/cmake/3.20.3/bin/cmake.exe -P C:/Users/acer/Desktop/esp-idf/tools/cmake/scripts/fail.cmake"
Failed to create SPIFFS image for partition 'model'.  Check project configuration if using the correct partition table file.
CMake Error at C:/Users/acer/Desktop/esp-idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

nikhil-robinson, 2021-12-07:
@TiramisuJ I tried hello world program with the sdkconfig of esp who and it worked fine

yehangyang, 2021-12-07:
@nikhil-robinson are you using ESP-S3-DevKit ?

TiramisuJ, 2021-12-07:
Oh... It doesn't make sense. your project crashed in the cpu_start stage，but in hello world, the cpu_start stage finished successfully. Can you run the [example](https://github.com/espressif/esp-dl/tree/master/examples/human_face_detect) in esp-dl.

nikhil-robinson, 2021-12-07:
@yehangyang  No i am using custom pcb for esp32s3

nikhil-robinson, 2021-12-07:
@TiramisuJ  i tried the human face detection of esp-DL and it works fine

output  : 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1858
load:0x403b6000,len:0xb74
load:0x403ba000,len:0x3014
entry 0x403b61ec
I (25) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (25) boot: compile time 13:45:25
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00200000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=f1a5ch (989788) map
I (256) esp_image: segment 1: paddr=00101a84 vaddr=3fc912f0 size=02900h ( 10496) load
I (258) esp_image: segment 2: paddr=0010438c vaddr=40374000 size=0bc8ch ( 48268) load
I (271) esp_image: segment 3: paddr=00110020 vaddr=42000020 size=6f8a8h (456872) map
I (341) esp_image: segment 4: paddr=0017f8d0 vaddr=4037fc8c size=01658h (  5720) load
I (342) esp_image: segment 5: paddr=00180f30 vaddr=50000000 size=00010h (    16) load
I (352) boot: Loaded app from partition at offset 0x10000
I (353) boot: Disabling RNG early entropy source...
W (358) bootloader_random: RNG for ESP32-S3 not currently supported
I (376) spiram: Found 16MBit SPI RAM device
I (377) spiram: SPI RAM mode: sram 80m
I (377) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (381) cpu_start: Pro cpu up.
I (385) cpu_start: Starting app cpu, entry point is 0x4037524c
0x4037524c: call_start_cpu1 at C:/Users/acer/Desktop/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (585) spiram: SPI SRAM memory test OK
I (594) cpu_start: Pro cpu start user code
I (594) cpu_start: cpu freq: 240000000
I (594) cpu_start: Application information:
I (597) cpu_start: Project name:     human_face_detect
I (603) cpu_start: App version:      v1.1.0-1-gd141502-dirty
I (609) cpu_start: Compile time:     Dec  7 2021 13:45:08
I (615) cpu_start: ELF file SHA256:  430bd4979409face...
I (621) cpu_start: ESP-IDF:          v4.4-dev-3703-gddc44956bf
I (628) heap_init: Initializing. RAM available for dynamic allocation:
I (635) heap_init: At 3FC96C28 len 000493D8 (292 KiB): D/IRAM
I (641) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (648) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (654) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (663) spi_flash: detected chip: generic
I (667) spi_flash: flash io: qio
W (671) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (689) sleep: Configure to isolate all GPIO pins in sleep state
I (691) sleep: Enable automatic switching of GPIO sleep configuration
I (698) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (718) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
Inference latency:           56571 us
[0] score: 0.987812, box: [137, 75, 246, 215]
    left eye: (157, 131), right eye: (199, 133)
    nose: (170, 163)
    mouth left: (158, 177), mouth right: (193, 180)

TiramisuJ, 2021-12-07:
Ok, Maybe your problem is caused by camera, In order to confirm this, you can delete the camera task in esp-who/examples/human_face_detection/terminal/main/app_main.cpp line 10, and run the esp-who again.

nikhil-robinson, 2021-12-07:
@TiramisuJ  No luck still showing same issue

TiramisuJ, 2021-12-21:
Hi @nikhil-robinson If you haven't solved the problem yet, I think maybe you can try this method:
Step 1: idf.py menuconfig → Component config → ESP System Settings → Channel for console output . set it to UART0
Step 2: idf.py menuconfig → Serial flasher config → Disable download stub. turn it off.
Then build and flash again. I hope this offer will be helpful to you.

nikhil-robinson, 2021-12-21:
Hi @TiramisuJ  I have tried this now i am getting this output

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1964
load:0x403b6000,len:0xc3c
load:0x403ba000,len:0x3014
entry 0x403b6218
I (25) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (25) boot: compile time 11:15:57
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 factory          factory app      00 00 00010000 003c0000
I (78) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (86) boot:  2 fr               unknown          20 20 003e0000 00020000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=5fd9ch (392604) map
I (165) esp_image: segment 1: paddr=0006fdc4 vaddr=3fc92b90 size=00254h (   596) load
I (166) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=7b488h (504968) map
I (248) esp_image: segment 3: paddr=000eb4b0 vaddr=3fc92de4 size=03770h ( 14192) load
I (250) esp_image: segment 4: paddr=000eec28 vaddr=40374000 size=0eb84h ( 60292) load
I (265) esp_image: segment 5: paddr=000fd7b4 vaddr=50000000 size=00010h (    16) load
I (271) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
W (274) bootloader_random: RNG for ESP32-S3 not currently supported
I (292) spiram: Found 16MBit SPI RAM device
I (292) spiram: SPI RAM mode: sram 80m
I (292) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (297) cpu_start: Pro cpu up.
I (301) cpu_start: Starting app cpu, entry point is 0x40375380
0x40375380: call_start_cpu1 at C:/Users/acer/Desktop/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (501) spiram: SPI SRAM memory test OK
I (510) cpu_start: Pro cpu start user code
I (510) cpu_start: cpu freq: 240000000
I (510) cpu_start: Application information:
I (513) cpu_start: Project name:     human_face_detection_terminal
I (520) cpu_start: App version:      v0.9.3-140-gbb0357d-dirty
I (526) cpu_start: Compile time:     Dec 21 2021 11:15:45
I (532) cpu_start: ELF file SHA256:  2b7df945715d32f9...
I (538) cpu_start: ESP-IDF:          v4.4-dev-3703-gddc44956bf
I (545) heap_init: Initializing. RAM available for dynamic allocation:
I (552) heap_init: At 3FC995C8 len 00046A38 (282 KiB): D/IRAM
I (558) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (565) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (571) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (580) spi_flash: detected chip: generic
I (584) spi_flash: flash io: qio
I (593) sleep: Configure to isolate all GPIO pins in sleep state
I (595) sleep: Enable automatic switching of GPIO sleep configuration
I (602) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (632) spiram: Res�

TiramisuJ, 2021-12-21:
Still not working properly? Check the configurations of Camera Pins and PSRAM.

nikhil-robinson, 2021-12-27:
HI @TiramisuJ The camera pin configuration and PSRAM is correc the main function is not being called at all

yan-qi-2009, 2022-01-14:
Hi @nikhil-robinson i got the same problem as you,did you solved the problem?

nikhil-robinson, 2022-01-15:
Hi @yan-qi-2009  No the issue still persist. Are you using esp32s3 chip?

ESP-YHY, 2022-03-07:
Hi @nikhil-robinson and @yan-qi-2009, has this problem been solved?

closed 2022-03-14 by Auroragan