espressif/esp-who #189: esp-idf 5.0: periph_module_enable() not found

pius-braun, 2021-12-03:
Thank you all for this great repository.

I just installed ESP-IDF 5.0 and then compiled the latest version of "examples/human_face_detection/terminal",
prompting a compiler error "implicit declaration of function".
With esp-idf 4.3, it compiled successfully.

I'm on WIndows 10, my target is the AI-Thinker esp32-cam.
This is a snippet of the error messages.
```
...
C:/ESP32/esp-who/components/esp32-camera/target/esp32/ll_cam.c:281:5: error: implicit declaration of function 'periph_module_enable'; did you mean 'esp_intr_enable'? [-Werror=implicit-function-declaration]
     periph_module_enable(PERIPH_I2S0_MODULE);
     ^~~~~~~~~~~~~~~~~~~~
...
C:/ESP32/esp-who/components/esp32-camera/target/esp32/ll_cam.c:281:26: error: 'PERIPH_I2S0_MODULE' undeclared (first use in this function); did you mean 'PERIPHS_IO_MUX_MTDO_U'?
     periph_module_enable(PERIPH_I2S0_MODULE);
                          ^~~~~~~~~~~~~~~~~~
...
C:/ESP32/esp-who/components/bus/i2s_lcd_esp32_driver.c:438:9: error: implicit declaration of function 'periph_module_enable'; did you mean 'i2s_adc_enable'? [-Werror=implicit-function-declaration]
         periph_module_enable(PERIPH_I2S0_MODULE);
         ^~~~~~~~~~~~~~~~~~~~
```

My workaround:
I added to C:\ESP32\esp-who\components\esp32-camera\target\esp32\ll_cam.c 
```
#include "esp_private/periph_ctrl.h"
#include "soc/periph_defs.h"
```
and to C:\ESP32\esp-who\components\bus\i2s_lcd_esp32_driver.c

`#include "esp_private/periph_ctrl.h"`

It does look wrong, to include from an "esp_private" directory. Any other fix?

yehangyang, 2021-12-09:
@pius-braun It seems the ESP-IDF change some API names in the master branch, the esp-idf 5.0. Would you mind switching esp-idf to v4.4-dev?

pius-braun, 2021-12-09:
It worked with v4.4.
```
C:\ESP32\esp-who\examples\human_face_detection\terminal>idf.py --version
ESP-IDF v4.4-beta1
```
Thanks.

closed 2022-02-25 by Auroragan

eleclong, 2023-08-08:
add esp_hw_support in cmakelist