espressif/esp-who #192: ESP EYE - Configure a camera

TomerMichaelovich, 2021-12-11:
Hi, everyone

I have ESP -EYE.
In the meantime, I was only able to run the Hello_World example.

When I want to run an example that requires a camera I can not set the camera

In the settings menu I do not have a camera setting.
And gets error:

[image]

This is what my settings menu looks like

[image]

ESP-WHO does not appear

Would appreciate help

yehangyang, 2021-12-13:
@TomerMichaelovich According to your path, 'C:\esp\human_face_detection\lcd', it seems you moved the example out of ESP-WHO. That makes some components missing. Please git clone the whole ESP-WHO including submodules. Then try the example again.

TomerMichaelovich, 2021-12-13:
Thank you very much !

When you say submodels you mean this command

git submodule update --recursive --init  ?

I did what you said and now get another error

[image]

TomerMichaelovich, 2021-12-13:
Thank you very much !

It works ! I got the following screen:

[image]

How can I see the video received from the camera?

Thanks

yehangyang, 2021-12-30:
@TomerMichaelovich It seems you are using ESP-EYE(ESP32) right? We'll release a new example with a web server, then you can see the video from the camera.

Auroragan, 2022-02-25:
Hi, the new example with web server has been updated: https://github.com/espressif/esp-who/tree/master/examples/human_face_detection/web, please try it out.

closed 2022-02-25 by Auroragan