espressif/esp-who #194: Makefile:129: recipe for target 'all' failed

We-Sky, 2022-01-08:
~/esp-who/examples/human_face_detection/lcd和~/esp-who/examples/human_face_recognition/lcd build的时候都提示这个报错编译失败

ESP-YHY, 2022-01-13:
请提供更多的 log 信息，谢谢

We-Sky, 2022-01-14:
> 请提供更多的 log 信息，谢谢
Executing action: all (aliases: build)
Running make in directory /home/zhongtian/esp-who/examples/single_chip/camera_web_server/build
Executing "make -j 14 all"...
[  0%] Built target memory_ld
[  0%] Built target custom_bundle
[  0%] Built target _project_elf_src
[  1%] Built target partition_table_bin
[  1%] Performing build step for 'bootloader'
[  1%] Built target __idf_ulp
[  1%] Built target __idf_esp_serial_slave_link
[  1%] Built target _project_elf_src
[  5%] Built target __idf_log
[  2%] Built target __idf_sdmmc
[ 11%] Built target __idf_esp_rom
[  2%] Built target __idf_esp_https_ota
[ 13%] Built target __idf_esp_common
[  3%] Built target __idf_esp_http_server
[ 16%] Built target __idf_xtensa
[  4%] Built target __idf_esp_http_client
[ 27%] Built target __idf_esp_hw_support
[  5%] Built target __idf_tcp_transport
[ 29%] Built target __idf_esp_system
[  5%] Built target __idf_esp-tls
[ 38%] Built target __idf_efuse
[  7%] Built target __idf_nghttp
[ 62%] Built target __idf_bootloader_support
[  7%] Built target __idf_cxx
[ 64%] Built target __idf_spi_flash
[  9%] Built target __idf_newlib
[ 67%] Built target __idf_micro-ecc
[ 11%] Built target __idf_freertos
[ 89%] Built target __idf_soc
[ 12%] Built target __idf_esp_timer
[ 93%] Built target __idf_hal
[ 95%] Built target __idf_main
[ 12%] Built target __idf_esp_common
[ 98%] Built target bootloader.elf
[ 13%] Built target __idf_xtensa
[100%] Built target gen_project_binary
[ 16%] Built target __idf_esp_hw_support
Bootloader binary size 0x6110 bytes. 0xef0 bytes (13%) free.
[100%] Built target bootloader_check_size
[ 18%] Built target __idf_soc
[100%] Built target app
[ 19%] Built target __idf_heap
[ 19%] No install step for 'bootloader'
[ 19%] Built target __idf_log
[ 19%] Completed 'bootloader'
[ 20%] Built target bootloader
[ 30%] Built target __idf_lwip
[ 31%] Built target __idf_esp_wifi
[ 39%] Built target __idf_wpa_supplicant
[ 39%] Built target __idf_esp_event
[ 40%] Built target __idf_esp_netif
[ 40%] Built target __idf_tcpip_adapter
[ 42%] Built target __idf_esp_eth
[ 42%] Built target __idf_vfs
[ 47%] Built target __idf_hal
[ 48%] Built target __idf_esp_rom
[ 52%] Built target __idf_esp_system
[ 52%] Built target __idf_esp_phy
[ 53%] Built target __idf_pthread
[ 55%] Built target __idf_nvs_flash
[ 57%] Built target __idf_spi_flash
[ 58%] Built target __idf_bootloader_support
[ 58%] Built target __idf_app_update
[ 66%] Built target mbedcrypto
[ 67%] Built target mbedx509
[ 68%] Built target mbedtls
[... 78 lines cut ...]
cc1: some warnings being treated as errors
esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/build.make:134: recipe for target 'esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/xclk.c.obj' failed
make[2]: *** [esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/xclk.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:5903: recipe for target 'esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/all' failed
make[1]: *** [esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 98%] Linking C static library libesp_local_ctrl.a
[ 98%] Built target __idf_esp_local_ctrl
[ 98%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/wifi_config.c.obj
[ 98%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/wifi_scan.c.obj
[ 98%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/handlers.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_console.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_scan.pb-c.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_config.pb-c.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_softap.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_constants.pb-c.c.obj
[ 99%] Linking C static library libwifi_provisioning.a
[ 99%] Built target __idf_wifi_provisioning
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

这是我的编译过程输出

ESP-YHY, 2022-01-14:
> ts/esp32-camera/driver/xclk.c:34:26: error: 'PERIPH_LEDC_MO

请使用 IDF 的 release/v4.4 试一下。

We-Sky, 2022-01-14:
> > ts/esp32-camera/driver/xclk.c:34:26: error: 'PERIPH_LEDC_MO
> 
> 请使用 IDF 的 release/v4.4 试一下。

我看esp-who的文件夹下也有一个esp-idf，是需要替换这个吗，还是把之前设置的esp-idf重新下载一个

ESP-YHY, 2022-01-14:
> > > ts/esp32-camera/driver/xclk.c:34:26: error: 'PERIPH_LEDC_MO
> > 
> > 
> > 请使用 IDF 的 release/v4.4 试一下。
> 
> 我看esp-who的文件夹下也有一个esp-idf，是需要替换这个吗，还是把之前设置的esp-idf重新下载一个

最新的 esp-who 的文件夹下应该是没有 esp-idf 的。你使用哪个 IDF 就更新哪个 IDF 就行。

We-Sky, 2022-01-14:
> > > > ts/esp32-camera/driver/xclk.c:34:26: error: 'PERIPH_LEDC_MO
> > > 
> > > 
> > > 请使用 IDF 的 release/v4.4 试一下。
> > 
> > 
> > 我看esp-who的文件夹下也有一个esp-idf，是需要替换这个吗，还是把之前设置的esp-idf重新下载一个
> 
> 最新的 esp-who 的文件夹下应该是没有 esp-idf 的。你使用哪个 IDF 就更新哪个 IDF 就行。

更新是直接clone 4.4版本还是有别的命令呀

We-Sky, 2022-01-14:
你好 我重新装了一遍idf 4.4后还是报错呀

[ 74%] Building C object esp-idf/spiffs/CMakeFiles/__idf_spiffs.dir/spiffs/src/spiffs_cache.c.obj
[ 74%] Building C object esp-idf/openssl/CMakeFiles/__idf_openssl.dir/platform/ssl_pm.c.obj
[ 74%] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_io_i80.c.obj
[ 74%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_auth/crypto_auth.c.obj
[ 74%] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cbortojson.c.obj
[ 75%] Building C object esp-idf/esp-face/CMakeFiles/__idf_esp-face.dir/face_detection/fd_forward.c.obj
[ 75%] Built target __idf_cmock
[ 75%] Building C object esp-idf/spiffs/CMakeFiles/__idf_spiffs.dir/spiffs/src/spiffs_check.c.obj
[ 75%] Building C object esp-idf/spiffs/CMakeFiles/__idf_spiffs.dir/spiffs/src/spiffs_gc.c.obj
[ 75%] Linking C static library libasio.a
[ 75%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_auth/hmacsha256/auth_hmacsha256.c.obj
[ 76%] Linking C static library libmqtt.a
[ 76%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/src/common/protocomm.c.obj
[ 76%] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_nt35510.c.obj
[ 76%] Building C object esp-idf/openssl/CMakeFiles/__idf_openssl.dir/platform/ssl_port.c.obj
[ 76%] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborvalidation.c.obj
[ 76%] Building C object esp-idf/spiffs/CMakeFiles/__idf_spiffs.dir/spiffs/src/spiffs_hydrogen.c.obj
[ 76%] Building CXX object esp-idf/esp-face/CMakeFiles/__idf_esp-face.dir/object_detection/object_detection.cpp.obj
[ 76%] Building C object esp-idf/spiffs/CMakeFiles/__idf_spiffs.dir/spiffs/src/spiffs_nucleus.c.obj
[ 76%] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/camera.c.obj
[ 76%] Linking C static library libopenssl.a
[ 76%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_auth/hmacsha512/auth_hmacsha512.c.obj
[ 76%] Built target __idf_asio
[ 76%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/src/security/security0.c.obj
[ 76%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/src/security/security1.c.obj
[ 76%] Built target __idf_mqtt
[ 76%] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/open_memstream.c.obj
[ 77%] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_ssd1306.c.obj
[ 77%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/proto-c/constants.pb-c.c.obj
[ 77%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/proto-c/sec0.pb-c.c.obj
[ 77%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/proto-c/session.pb-c.c.obj
[ 77%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/proto-c/sec1.pb-c.c.obj
[ 77%] Linking C static library libspiffs.a
[ 77%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_auth/hmacsha512256/auth_hmacsha512256.c.obj
[ 77%] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_st7789.c.obj
[ 77%] Linking C static library libcbor.a
[ 77%] Building C object esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/address.c.obj
[ 77%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/src/transports/protocomm_console.c.obj
[ 77%] Building C object esp-idf/esp-face/CMakeFiles/__idf_esp-face.dir/face_recognition/fr_forward.c.obj
[ 77%] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/sccb.c.obj
[ 77%] Built target __idf_openssl
[ 78%] Building C object esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/async.c.obj
[ 78%] Building C object esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/block.c.obj
[ 78%] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/src/transports/protocomm_httpd.c.obj
[ 78%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/crypto_box.c.obj
[ 78%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/crypto_box_easy.c.obj
[ 78%] Built target __idf_spiffs
[ 79%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/crypto_box_seal.c.obj
[ 79%] Built target __idf_cbor
[ 79%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.c.obj
[ 79%] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_ops.c.obj
[ 79%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj
[ 80%] Building C object esp-idf/fatfs/CMakeFiles/__idf_fatfs.dir/diskio/diskio.c.obj
[ 80%] Building C object esp-idf/esp-face/CMakeFiles/__idf_esp-face.dir/pose_estimation/pe_forward.c.obj
[ 80%] Building C object esp-idf/esp-face/CMakeFiles/__idf_esp-face.dir/face_recognition/fr_flash.c.obj
[ 80%] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.c.obj
[ 80%] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/sensor.c.obj
[... cut ...]
esp-idf/esp_system/CMakeFiles/__ldgen_output_sections.ld.dir/build.make:163: recipe for target 'esp-idf/esp_system/ld/sections.ld' failed
make[2]: *** [esp-idf/esp_system/ld/sections.ld] Error 1
CMakeFiles/Makefile2:2679: recipe for target 'esp-idf/esp_system/CMakeFiles/__ldgen_output_sections.ld.dir/all' failed
make[1]: *** [esp-idf/esp_system/CMakeFiles/__ldgen_output_sections.ld.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

ESP-YHY, 2022-01-14:
> 我发现我按照文档克隆后执行`git submodule update --recursive --init` 得到的是 [image] 这个版本 但是master分支里面没有camera_web的例程，是因为这个问题报错的吗

ESP-WHO 是最新的吗？

We-Sky, 2022-02-11:
> ESP-IDF已经更新到最新，ESP-WHO也是更新到最新了，还是报错呢
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

Auroragan, 2022-02-25:
你好，该问题已修复，请重新尝试。

closed 2022-02-25 by Auroragan