espressif/esp-who #197: build the ESP-WHO example project and get error like this

yan-qi-2009, 2022-01-14:
C:\ESP_IDF\esp-idf\esp-who\examples\human_face_detection\terminal>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp_idf\esp-idf\esp-who\examples\human_face_detection\terminal\build
Executing "ninja all"...
[3/1105] Running move model...
'rm' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
'cp' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
'cp' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
C:/ESP_IDF/esp-idf/esp-who/examples/human_face_detection/terminal/sdkconfig
C:/ESP_IDF/esp-idf/esp-who/components/esp-sr/model
C:/ESP_IDF/esp-idf/esp-who/components/esp-sr/model/wakenet_model/alexa8
C:/ESP_IDF/esp-idf/esp-who/components/esp-sr/model/multinet_model/mn3cn
Recommended model partition size:  500KB
[5/1105] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
factory,app,factory,0x10000,3840K,
nvs,data,nvs,0x3d0000,16K,
fr,32,32,0x3e0000,128K,
*******************************************************************************
[7/1105] cmd.exe /C "cd /D C:\ESP_IDF\esp-idf\esp-who\exam....exe -P C:/ESP_IDF/esp-idf/tools/cmake/scripts/fail.cmake"
FAILED: esp-idf/esp-sr/CMakeFiles/spiffs_model_bin
cmd.exe /C "cd /D C:\ESP_IDF\esp-idf\esp-who\examples\human_face_detection\terminal\build\esp-idf\esp-sr && C:\Users\THS-QIYAN\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E echo "Failed to create SPIFFS image for partition 'model'. " "Check project configuration if using the correct partition table file." && C:\Users\THS-QIYAN\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E remove C:/ESP_IDF/esp-idf/esp-who/examples/human_face_detection/terminal/build/esp-idf/esp-sr/izFYu.cmake && C:\Users\THS-QIYAN\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" C:/Users/THS-QIYAN/.espressif/tools/cmake/3.20.3/bin/cmake.exe -P C:/ESP_IDF/esp-idf/tools/cmake/scripts/fail.cmake"
Failed to create SPIFFS image for partition 'model'.  Check project configuration if using the correct partition table file.
CMake Error at C:/ESP_IDF/esp-idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

ESP-YHY, 2022-01-18:
Hi @yan-qi-2009, 请问你在什么环境下运行的?
请使用 IDF/release/v4.4 进行编译。

yan-qi-2009, 2022-01-20:
就是在v4.4进行编译的

yan-qi-2009, 2022-01-20:
@ESP-YHY 就是在v4.4进行编译的

ESP-YHY, 2022-01-20:
@yan-qi-2009 是在 Windows 下吗？

yan-qi-2009, 2022-01-20:
@ESP-YHY 是的

ESP-YHY, 2022-01-20:
@yan-qi-2009, 我刚刚在 Windows 7 下尝试了，可以正常编译通过。
你可以正常编译 IDF 的示例吗？

Beanv587, 2022-01-30:
@ESP-YHY 
工程默认是esp32,当然可以编译了.你先设置一下编译目标: idf.py set-target esp32s3 然后再idf.py build 就不行了,我这也跟楼主一样的情况

We-Sky, 2022-02-12:
请问你解决了吗？我也遇到了这个问题

ESP-YHY, 2022-02-14:
@yan-qi-2009 请暂时参考[#201](https://github.com/espressif/esp-who/issues/201)解决问题。我们会尽快修复bug。

Auroragan, 2022-02-25:
你好，该问题已修复，请重新尝试。

closed 2022-02-25 by Auroragan