espressif/esp-who #201: [esp32-s3] 无法编译任何一个示例

Beanv587, 2022-01-30:
问题重现步骤:
1.进入任意一个示例目录,比如/examples/human_face_detection/lcd
2.设置编译目标: idf.py set-target esp32s3
3.编译: idf.py build

linux(ubuntu)、windows(win10) 都存在这个问题.如果采用默认esp32是可以编译的

linux截图如下:
[image]

windows跟issuse里的AIV-491的情况一模一样

Beanv587, 2022-02-01:
自己解决了,要把sdkconfig.defaults.esp32s3改成sdkconfig.defaults.

stanley-sz, 2022-02-07:
我也是遇到同样的问题，楼主解决了没呢？谢谢！

stanley-sz, 2022-02-07:
已解决，在esp-who/components/esp-sr 中 git checkout master

yehangyang, 2022-02-07:
你好@Beanv587， 请尝试在仓库的根目录下 `git submodule update --init`，看这样能否解决问题。

Beanv587, 2022-02-07:
> 已解决，在esp-who/components/esp-sr 中 git checkout master

@stanley-sz 谢谢,按你的方案我也成功编译了.看来还是有点坑...

@yehangyang git submodule update --init执行后没用,因为本身全部都是最新的

closed 2022-02-07 by Beanv587

ESP-YHY, 2022-02-14:
@Beanv587 感谢，我会尽快更新 esp-sr 的commit号，再次感谢发现问题。