espressif/esp-who #204: esp-who 编译失败

We-Sky, 2022-02-12:
使用的vscode插件编译，blink程序可以编译成功，ESP-IDF版本是release v4.4，提示以下信息：
>
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/sdkconfig
Loading defaults file D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/sdkconfig.defaults...
Loading defaults file D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/sdkconfig.defaults.esp32s3...
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "human_face_detection_lcd" version: v0.9.3-145-g3f9d35f-dirty
-- Adding linker script D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Users/Zhongtian/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt bus cbor cmock coap console cxx driver efuse esp-code-scanner esp-dl esp-sr esp-tls esp32-camera esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs fb_gfx freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns modules mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread screen sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa        
-- Component paths: C:/Users/Zhongtian/esp/esp-idf/components/app_trace C:/Users/Zhongtian/esp/esp-idf/components/app_update C:/Users/Zhongtian/esp/esp-idf/components/asio C:/Users/Zhongtian/esp/esp-idf/components/bootloader C:/Users/Zhongtian/esp/esp-idf/components/bootloader_support C:/Users/Zhongtian/esp/esp-idf/components/bt D:/OneDrive/code/ESP32/esp-who/components/bus C:/Users/Zhongtian/esp/esp-idf/components/cbor C:/Users/Zhongtian/esp/esp-idf/components/cmock C:/Users/Zhongtian/esp/esp-idf/components/coap C:/Users/Zhongtian/esp/esp-idf/components/console C:/Users/Zhongtian/esp/esp-idf/components/cxx C:/Users/Zhongtian/esp/esp-idf/components/driver C:/Users/Zhongtian/esp/esp-idf/components/efuse D:/OneDrive/code/ESP32/esp-who/components/esp-code-scanner D:/OneDrive/code/ESP32/esp-who/components/esp-dl D:/OneDrive/code/ESP32/esp-who/components/esp-sr C:/Users/Zhongtian/esp/esp-idf/components/esp-tls D:/OneDrive/code/ESP32/esp-who/components/esp32-camera C:/Users/Zhongtian/esp/esp-idf/components/esp32s3 C:/Users/Zhongtian/esp/esp-idf/components/esp_adc_cal C:/Users/Zhongtian/esp/esp-idf/components/esp_common C:/Users/Zhongtian/esp/esp-idf/components/esp_eth C:/Users/Zhongtian/esp/esp-idf/components/esp_event C:/Users/Zhongtian/esp/esp-idf/components/esp_gdbstub C:/Users/Zhongtian/esp/esp-idf/components/esp_hid C:/Users/Zhongtian/esp/esp-idf/components/esp_http_client C:/Users/Zhongtian/esp/esp-idf/components/esp_http_server C:/Users/Zhongtian/esp/esp-idf/components/esp_https_ota C:/Users/Zhongtian/esp/esp-idf/components/esp_https_server C:/Users/Zhongtian/esp/esp-idf/components/esp_hw_support C:/Users/Zhongtian/esp/esp-idf/components/esp_ipc C:/Users/Zhongtian/esp/esp-idf/components/esp_lcd C:/Users/Zhongtian/esp/esp-idf/components/esp_local_ctrl C:/Users/Zhongtian/esp/esp-idf/components/esp_netif C:/Users/Zhongtian/esp/esp-idf/components/esp_phy C:/Users/Zhongtian/esp/esp-idf/components/esp_pm C:/Users/Zhongtian/esp/esp-idf/components/esp_ringbuf C:/Users/Zhongtian/esp/esp-idf/components/esp_rom C:/Users/Zhongtian/esp/esp-idf/components/esp_serial_slave_link C:/Users/Zhongtian/esp/esp-idf/components/esp_system C:/Users/Zhongtian/esp/esp-idf/components/esp_timer C:/Users/Zhongtian/esp/esp-idf/components/esp_websocket_client C:/Users/Zhongtian/esp/esp-idf/components/esp_wifi C:/Users/Zhongtian/esp/esp-idf/components/espcoredump C:/Users/Zhongtian/esp/esp-idf/components/esptool_py C:/Users/Zhongtian/esp/esp-idf/components/expat C:/Users/Zhongtian/esp/esp-idf/components/fatfs D:/OneDrive/code/ESP32/esp-who/components/fb_gfx C:/Users/Zhongtian/esp/esp-idf/components/freemodbus C:/Users/Zhongtian/esp/esp-idf/components/freertos C:/Users/Zhongtian/esp/esp-idf/components/hal C:/Users/Zhongtian/esp/esp-idf/components/heap C:/Users/Zhongtian/esp/esp-idf/components/idf_test C:/Users/Zhongtian/esp/esp-idf/components/ieee802154 C:/Users/Zhongtian/esp/esp-idf/components/jsmn C:/Users/Zhongtian/esp/esp-idf/components/json C:/Users/Zhongtian/esp/esp-idf/components/libsodium C:/Users/Zhongtian/esp/esp-idf/components/log C:/Users/Zhongtian/esp/esp-idf/components/lwip D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/main C:/Users/Zhongtian/esp/esp-idf/components/mbedtls C:/Users/Zhongtian/esp/esp-idf/components/mdns D:/OneDrive/code/ESP32/esp-who/components/modules C:/Users/Zhongtian/esp/esp-idf/components/mqtt C:/Users/Zhongtian/esp/esp-idf/components/newlib C:/Users/Zhongtian/esp/esp-idf/components/nghttp C:/Users/Zhongtian/esp/esp-idf/components/nvs_flash C:/Users/Zhongtian/esp/esp-idf/components/openssl C:/Users/Zhongtian/esp/esp-idf/components/openthread C:/Users/Zhongtian/esp/esp-idf/components/partition_table C:/Users/Zhongtian/esp/esp-idf/components/perfmon C:/Users/Zhongtian/esp/esp-idf/components/protobuf-c C:/Users/Zhongtian/esp/esp-idf/components/protocomm C:/Users/Zhongtian/esp/esp-idf/components/pthread D:/OneDrive/code/ESP32/esp-who/components/screen C:/Users/Zhongtian/esp/esp-idf/components/sdmmc C:/Users/Zhongtian/esp/esp-idf/components/soc C:/Users/Zhongtian/esp/esp-idf/components/spi_flash C:/Users/Zhongtian/esp/esp-idf/components/spiffs C:/Users/Zhongtian/esp/esp-idf/components/tcp_transport C:/Users/Zhongtian/esp/esp-idf/components/tcpip_adapter C:/Users/Zhongtian/esp/esp-idf/components/tinyusb C:/Users/Zhongtian/esp/esp-idf/components/ulp C:/Users/Zhongtian/esp/esp-idf/components/unity C:/Users/Zhongtian/esp/esp-idf/components/usb C:/Users/Zhongtian/esp/esp-idf/components/vfs C:/Users/Zhongtian/esp/esp-idf/components/wear_levelling C:/Users/Zhongtian/esp/esp-idf/components/wifi_provisioning C:/Users/Zhongtian/esp/esp-idf/components/wpa_supplicant C:/Users/Zhongtian/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/build
[585/1079] cmd.exe /C "cd /D D:\OneDrive\code\ESP32\esp-who\examples\h....exe -P C:/Users/Zhongtian/esp/esp-idf/tools/cmake/scripts/fail.cmake" 
FAILED: esp-idf/esp-sr/CMakeFiles/spiffs_model_bin
cmd.exe /C "cd /D D:\OneDrive\code\ESP32\esp-who\examples\human_face_detection\lcd\build\esp-idf\esp-sr && C:\Users\Zhongtian\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E echo "Failed to create SPIFFS image for partition 'model'. " "Check project configuration if using the correct partition table file." && C:\Users\Zhongtian\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E remove D:/OneDrive/code/ESP32/esp-who/examples/human_face_detection/lcd/build/esp-idf/esp-sr/cOMUR.cmake && C:\Users\Zhongtian\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" C:/Users/Zhongtian/.espressif/tools/cmake/3.20.3/bin/cmake.exe -P C:/Users/Zhongtian/esp/esp-idf/tools/cmake/scripts/fail.cmake"
Failed to create SPIFFS image for partition 'model'.  Check project configuration if using the correct partition table file.
CMake Error at C:/Users/Zhongtian/esp/esp-idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

[587/1079] Running move model...
'rm' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
[... cut ...]
[16/99] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/lcd_periph.c.obj
[17/99] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/rtc_io_periph.c.obj
[18/99] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/wdt_hal_iram.c.obj
[19/99] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/ledc_periph.c.obj
[20/99] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/sdmmc_periph.c.obj
[21/99] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s3/uart_periph.c.obj

Beanv587, 2022-02-12:
解决方法在这里:https://github.com/espressif/esp-who/issues/201

在esp-who/components/esp-sr 中 git checkout master

We-Sky, 2022-02-13:
@Beanv587 感谢，已解决

closed 2022-02-25 by Auroragan