espressif/esp-who #212: Errors when building examples using IDF

castrorenam, 2022-04-10:
Hi, 

I am trying to build the example projects, but I am getting errors. For instance, when trying to build the motion_detection/terminal example the building process crashes and gives the message ->

.../esp-who/components/bus/include/i2s_lcd_driver.h:17:10: fatal error: driver/i2s.h: No such file or directory

I just cloned the git repo and I am sure nothing was deleted. I am following the instructions precisely as described.

jmadsenee, 2022-04-29:
I am having basically the same problem with another example.  That file (i2s.h) definitely exists.  It is in: 

...\esp-idf-master\components\driver\include\driver.  

The compiler is just unable to find it.  I have the IDF_PATH set to \esp-idf-master.  The thing that I don't understand is that it was compiling earlier, albeit with a pile of warnings like [-Wincompatible-pointer-types].  I had IDF V4.4.1 installed, so I tried installing IDF Master to hopefully fix the errors, and now I get this error.

Someone please help!

jmadsenee, 2022-04-29:
I am working in Windows 10 in a CMD window