espressif/esp-who #218: ESP32-S3-EYE Speaker support

ncohen17, 2022-06-15:
Hello, 

I recently received the ESP32-S3-EYE dev kit and I wanted to know if there is any way to add speaker output to the board as there is already a microphone on board so a speaker is the only missing component for my application use case.

liminghui-sandra, 2022-06-16:
Hello,
Unfortunately ESP-S3-EYE cannot be connected with a speaker because of limitation in HW board size. We'd suggest ESP32-S3-Korvo-2, which might meet your requirement of multiple peripherals.

ncohen17, 2022-06-16:
Hi @liminghui-sandra 

Thanks for your reply,
does the ESP32-S3-Korvo-2 also have camera support as I need camera, microphone and speaker support for my application.

ncohen17, 2022-06-16:
Where can I purchase one ? I don’t see that they are available for purchase.

liminghui-sandra, 2022-06-16:
Pls check mouser: https://www.mouser.hk/ProductDetail/Espressif-Systems/ESP32-S3-KORVO-2?qs=TCDPyi3sCW38YjcOnFwdyg%3D%3D

ncohen17, 2022-06-16:
Also which camera module is being used in the dev kit as I just read that the LCD and camera are sold separately.

liminghui-sandra, 2022-06-16:
the S3-Korvo-2-LCD is an optional component of S3-Korvo-2. Yes you may purchase it separately. It contains both LCD and camera （OV3660 by default）. 
 https://www.mouser.hk/ProductDetail/Espressif-Systems/ESP32-S3-KORVO-2-LCD?qs=Znm5pLBrcALHzum%2FtP9Ccg%3D%3D

closed 2023-02-24 by Auroragan