espressif/esp-who #219: doubt

prabhu012, 2022-06-16:
Sir I am adding this https://rarecomponents.com/store/1619 camera board in esp32s3 devkitC-1 board then doing  the face detection and recognition project 
But Here all the process are working properly but at the time of flash it is not connecting with wifi

this is configuration file
[costom camera pin and wifi.pdf](https://github.com/espressif/esp-who/files/8921230/costom.camera.pin.and.wifi.pdf)

[and when we flash we getting this]
[flash.txt](https://github.com/espressif/esp-who/files/8921233/flash.txt)

and when we monitor we getting this
[monitor.txt](https://github.com/espressif/esp-who/files/8921261/monitor.txt)

prabhu012, 2022-06-17:
tell me about this

 ()  Camera Host Name                                                                                                                                                   
 ()  WiFi STA SSID                                                                                                                                                     
 ()  WiFi STA Password                                                                                                                                                 
 (ESP32-Camera) WiFi AP SSID                                                                                                                                            
 ()  WiFi AP Password                                                                                                                                                   
 (1) Maximal STA connections                                                                                                                                          
 ()  WiFi AP Channel                                                                                                                                                    
 (192.168.4.1) WiFi AP IP Address                                                                                                                                        
 (5) Maximum retry

liminghui-sandra, 2022-06-17:
Hello,
pwdn and reset pins are usually not connected, these two pins should be configured to -1 in program.
Regarding wifi issue, pls check where the program is stuck. Suggest to verify by this example:  https://github.com/espressif/esp-iot-solution/tree/master/examples/camera

prabhu012, 2022-06-17:
in our program all the process working properly only our board is not connecting with wifi
and here board is work in AP Mode or STA Mode

prabhu012, 2022-06-17:
and tell me wifi configuration is Optional or not

prabhu012, 2022-06-23:
(760) MSPI Timing: tuning fail, best point is fallen back to index 1

sir I'm getting this error

Qitas, 2022-06-25:
Your esp32s3 devkitC-1 does not have PSRAM，but the demo you're running requires PSRAM (default 8MB OCT SPIRAM)

prabhu012, 2022-06-25:
then what next  what i take

prabhu012, 2022-06-27:
sir how too fixed this error

C:\esp-who\examples\human_face_detection\lcd>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp-who\examples\human_face_detection\lcd\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/esp-who/examples/human_face_detection/lcd/sdkconfig
Loading defaults file C:/esp-who/examples/human_face_detection/lcd/sdkconfig.defaults...
Loading defaults file C:/esp-who/examples/human_face_detection/lcd/sdkconfig.defaults.esp32s3...
C:/esp-who/examples/human_face_detection/lcd/sdkconfig.defaults.esp32s3:3 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
C:/esp-who/examples/human_face_detection/lcd/sdkconfig.defaults.esp32s3:4 CONFIG_ESP32S3_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
-- App "human_face_detection_lcd" version: v0.9.3-167-g7563e00
-- Adding linker script C:/esp-who/examples/human_face_detection/lcd/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt bus cmock console cxx driver efuse esp-code-scanner esp-dl esp-sr esp-tls esp32-camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py fatfs fb_gfx freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mdns modules mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread screen sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/esp/esp-idf/components/app_trace C:/esp/esp-idf/components/app_update C:/esp/esp-idf/components/asio C:/esp/esp-idf/components/bootloader C:/esp/esp-idf/components/bootloader_support C:/esp/esp-idf/components/bt C:/esp-who/components/bus C:/esp/esp-idf/components/cmock C:/esp/esp-idf/components/console C:/esp/esp-idf/components/cxx C:/esp/esp-idf/components/driver C:/esp/esp-idf/components/efuse C:/esp-who/components/esp-code-scanner C:/esp-who/components/esp-dl C:/esp-who/components/esp-sr C:/esp/esp-idf/components/esp-tls C:/esp-who/components/esp32-camera C:/esp/esp-idf/components/esp_adc_cal C:/esp/esp-idf/components/esp_common C:/esp/esp-idf/components/esp_eth C:/esp/esp-idf/components/esp_event C:/esp/esp-idf/components/esp_gdbstub C:/esp/esp-idf/components/esp_hid C:/esp/esp-idf/components/esp_http_client C:/esp/esp-idf/components/esp_http_server C:/esp/esp-idf/components/esp_https_ota C:/esp/esp-idf/components/esp_https_server C:/esp/esp-idf/components/esp_hw_support C:/esp/esp-idf/components/esp_lcd C:/esp/esp-idf/components/esp_local_ctrl C:/esp/esp-idf/components/esp_netif C:/esp/esp-idf/components/esp_phy C:/esp/esp-idf/components/esp_pm C:/esp/esp-idf/components/esp_psram C:/esp/esp-idf/components/esp_ringbuf C:/esp/esp-idf/components/esp_rom C:/esp/esp-idf/components/esp_serial_slave_link C:/esp/esp-idf/components/esp_system C:/esp/esp-idf/components/esp_timer C:/esp/esp-idf/components/esp_wifi C:/esp/esp-idf/components/espcoredump C:/esp/esp-idf/components/esptool_py C:/esp/esp-idf/components/fatfs C:/esp-who/components/fb_gfx C:/esp/esp-idf/components/freertos C:/esp/esp-idf/components/hal C:/esp/esp-idf/components/heap C:/esp/esp-idf/components/http_parser C:/esp/esp-idf/components/idf_test C:/esp/esp-idf/components/ieee802154 C:/esp/esp-idf/components/json C:/esp/esp-idf/components/log C:/esp/esp-idf/components/lwip C:/esp-who/examples/human_face_detection/lcd/main C:/esp/esp-idf/components/mbedtls C:/esp/esp-idf/components/mdns C:/esp-who/components/modules C:/esp/esp-idf/components/mqtt C:/esp/esp-idf/components/newlib C:/esp/esp-idf/components/nvs_flash C:/esp/esp-idf/components/openthread C:/esp/esp-idf/components/partition_table C:/esp/esp-idf/components/perfmon C:/esp/esp-idf/components/protobuf-c C:/esp/esp-idf/components/protocomm C:/esp/esp-idf/components/pthread C:/esp-who/components/screen C:/esp/esp-idf/components/sdmmc C:/esp/esp-idf/components/soc C:/esp/esp-idf/components/spi_flash C:/esp/esp-idf/components/spiffs C:/esp/esp-idf/components/tcp_transport C:/esp/esp-idf/components/tinyusb C:/esp/esp-idf/components/ulp C:/esp/esp-idf/components/unity C:/esp/esp-idf/components/usb C:/esp/esp-idf/components/vfs C:/esp/esp-idf/components/wear_levelling C:/esp/esp-idf/components/wifi_provisioning C:/esp/esp-idf/components/wpa_supplicant C:/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp-who/examples/human_face_detection/lcd/build
[... cut ...] ponents/bus/i2c_bus.c
In file included from C:/esp-who/components/bus/i2c_bus.c:21:
C:/esp-who/components/bus/include/i2c_bus.h:16:10: fatal error: driver/i2c.h: No such file or directory
   16 | #include "driver/i2c.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
[857/932] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/esp_camera.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\esp-who\examples\human_face_detection\lcd>

closed 2022-07-07 by prabhu012