espressif/esp-who #220: fatal error: driver/spi_master.h

absoluteabutaj, 2022-06-22:
Hello,

After cloning the `esp-who` repo and I tried to build the example `../../motion_detection/terminal`, it throws, 

```
mac@machine:~/esp/esp-who/examples/motion_detection/terminal$ idf.py flash monitor
Executing action: flash
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Running ninja in directory /home/mac/esp/esp-who/examples/motion_detection/terminal/build
Executing "ninja flash"...
[8/909] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[285/909] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mac/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mac/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mac/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/mac/esp/esp-who/examples/motion_detection/terminal/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Adding linker script /home/mac/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/mac/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/mac/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/mac/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/mac/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/mac/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/mac/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/mac/esp/esp-idf/components/bootloader /home/mac/esp/esp-idf/components/bootloader_support /home/mac/esp/esp-idf/components/efuse /home/mac/esp/esp-idf/components/esp_common /home/mac/esp/esp-idf/components/esp_hw_support /home/mac/esp/esp-idf/components/esp_rom /home/mac/esp/esp-idf/components/esp_system /home/mac/esp/esp-idf/components/esptool_py /home/mac/esp/esp-idf/components/freertos /home/mac/esp/esp-idf/components/hal /home/mac/esp/esp-idf/components/log /home/mac/esp/esp-idf/components/bootloader/subproject/main /home/mac/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /home/mac/esp/esp-idf/components/newlib /home/mac/esp/esp-idf/components/partition_table /home/mac/esp/esp-idf/components/soc /home/mac/esp/esp-idf/components/spi_flash /home/mac/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mac/esp/esp-who/examples/motion_detection/terminal/build/bootloader
[361/909] Performing build step for 'bootloader'
[1/100] Generating project_elf_src_esp32.c
[2/100] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/eri.c.obj
[3/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/adc_periph.c.obj
[4/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/dport_access_common.c.obj
[5/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[6/100] Building C object CMakeFiles/bootloader.elf.dir/project_elf_src_esp32.c.obj
[7/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/dport_access.c.obj
[8/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/interrupts.c.obj
[... 103 lines cut ...]
   16 | #include "driver/i2c.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
[826/909] Building C object esp-idf/bus/CM...s/__idf_bus.dir/i2s_lcd_esp32_driver.c.obj
FAILED: esp-idf/bus/CMakeFiles/__idf_bus.dir/i2s_lcd_esp32_driver.c.obj 
/home/mac/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/home/mac/esp/esp-who/examples/motion_detection/terminal/build/config -I/home/mac/esp/esp-who/components/bus/include -I/home/mac/esp/esp-idf/components/newlib/platform_include -I/home/mac/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/mac/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/mac/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/mac/esp/esp-idf/components/freertos/esp_additions/include -I/home/mac/esp/esp-idf/components/esp_hw_support/include -I/home/mac/esp/esp-idf/components/esp_hw_support/include/soc -I/home/mac/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/mac/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/mac/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/mac/esp/esp-idf/components/heap/include -I/home/mac/esp/esp-idf/components/log/include -I/home/mac/esp/esp-idf/components/soc/include -I/home/mac/esp/esp-idf/components/soc/esp32/. -I/home/mac/esp/esp-idf/components/soc/esp32/include -I/home/mac/esp/esp-idf/components/hal/esp32/include -I/home/mac/esp/esp-idf/components/hal/include -I/home/mac/esp/esp-idf/components/hal/platform_port/include -I/home/mac/esp/esp-idf/components/esp_rom/include -I/home/mac/esp/esp-idf/components/esp_rom/include/esp32 -I/home/mac/esp/esp-idf/components/esp_rom/esp32 -I/home/mac/esp/esp-idf/components/esp_common/include -I/home/mac/esp/esp-idf/components/esp_system/include -I/home/mac/esp/esp-idf/components/esp_system/port/soc -I/home/mac/esp/esp-idf/components/esp_system/port/include/private -I/home/mac/esp/esp-idf/components/xtensa/include -I/home/mac/esp/esp-idf/components/xtensa/esp32/include -I/home/mac/esp/esp-idf/components/lwip/include -I/home/mac/esp/esp-idf/components/lwip/include/apps -I/home/mac/esp/esp-idf/components/lwip/include/apps/sntp -I/home/mac/esp/esp-idf/components/lwip/lwip/src/include -I/home/mac/esp/esp-idf/components/lwip/port/esp32/include -I/home/mac/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/mac/esp/esp-idf/components/vfs/include -I/home/mac/esp/esp-idf/components/esp_wifi/include -I/home/mac/esp/esp-idf/components/esp_event/include -I/home/mac/esp/esp-idf/components/esp_netif/include -I/home/mac/esp/esp-idf/components/esp_phy/include -I/home/mac/esp/esp-idf/components/esp_phy/esp32/include -mlongcalls -Wno-frame-address  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=format= -Wno-format -Wno-enum-conversion -gdwarf-4 -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -fmacro-prefix-map=/home/mac/esp/esp-who/examples/motion_detection/terminal=. -fmacro-prefix-map=/home/mac/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-3654-gc2ccc383da-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/bus/CMakeFiles/__idf_bus.dir/i2s_lcd_esp32_driver.c.obj -MF esp-idf/bus/CMakeFiles/__idf_bus.dir/i2s_lcd_esp32_driver.c.obj.d -o esp-idf/ [... cut ...] df/bus/CMakeFiles/__idf_bus.dir/spi_bus.c.obj -c /home/mac/esp/esp-who/components/bus/spi_bus.c
/home/mac/esp/esp-who/components/bus/spi_bus.c:19:10: fatal error: driver/spi_master.h: No such file or directory
   19 | #include "driver/spi_master.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[830/909] Building C object esp-idf/esp-sr...f_esp-sr.dir/src/mn_process_commands.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

Am I missing something?

Auroragan, 2022-06-24:
hi, are you using ESP-IDF on the [release/v4.4](https://github.com/espressif/esp-idf/tree/release/v4.4) branch?

absoluteabutaj, 2022-06-24:
@Auroragan - I'm using 4.3.3

```
mac@machine:~$ idf.py --version
ESP-IDF v4.3.3-dirty
mac@machine:~$ 
```

`motion_detection` example built successfully, however when I try building `code_recognition`, this comes-up. I believe it is the compatibility issue with IDF. 

```
mac@machine:/media/mac/sec-vol/workspace-esp/code_recognition$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /media/mac/sec-vol/workspace-esp/code_recognition/build
Executing "ninja all"...
[2/12] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj 
/home/mac/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -I/media/mac/sec-vol/workspace-esp/code_recognition/build/config -I/media/mac/sec-vol/workspace-esp/code_recognition/main/include -I/home/mac/esp/esp-idf-v4.3.3/components/newlib/platform_include -I/home/mac/esp/esp-idf-v4.3.3/components/freertos/include -I/home/mac/esp/esp-idf-v4.3.3/components/freertos/port/xtensa/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_hw_support/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_hw_support/port/esp32/. -I/home/mac/esp/esp-idf-v4.3.3/components/esp_hw_support/port/esp32/private_include -I/home/mac/esp/esp-idf-v4.3.3/components/heap/include -I/home/mac/esp/esp-idf-v4.3.3/components/log/include -I/home/mac/esp/esp-idf-v4.3.3/components/lwip/include/apps -I/home/mac/esp/esp-idf-v4.3.3/components/lwip/include/apps/sntp -I/home/mac/esp/esp-idf-v4.3.3/components/lwip/lwip/src/include -I/home/mac/esp/esp-idf-v4.3.3/components/lwip/port/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/lwip/port/esp32/include/arch -I/home/mac/esp/esp-idf-v4.3.3/components/soc/include -I/home/mac/esp/esp-idf-v4.3.3/components/soc/esp32/. -I/home/mac/esp/esp-idf-v4.3.3/components/soc/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/hal/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/hal/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_rom/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_rom/esp32 -I/home/mac/esp/esp-idf-v4.3.3/components/esp_rom/include/esp32 -I/home/mac/esp/esp-idf-v4.3.3/components/esp_common/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_system/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/driver/include -I/home/mac/esp/esp-idf-v4.3.3/components/driver/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_ringbuf/include -I/home/mac/esp/esp-idf-v4.3.3/components/efuse/include -I/home/mac/esp/esp-idf-v4.3.3/components/efuse/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/xtensa/include -I/home/mac/esp/esp-idf-v4.3.3/components/xtensa/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/espcoredump/include -I/home/mac/esp/esp-idf-v4.3.3/components/espcoredump/include/port/xtensa -I/home/mac/esp/esp-idf-v4.3.3/components/esp_timer/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_ipc/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_pm/include -I/home/mac/esp/esp-idf-v4.3.3/components/vfs/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_wifi/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_wifi/esp32/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_event/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_netif/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_eth/include -I/home/mac/esp/esp-idf-v4.3.3/components/tcpip_adapter/include -I/home/mac/esp/esp-idf-v4.3.3/components/app_trace/include -I/home/mac/esp/esp-idf-v4.3.3/components/mbedtls/port/include -I/home/mac/esp/esp-idf-v4.3.3/components/mbedtls/mbedtls/include -I/home/mac/esp/esp-idf-v4.3.3/components/mbedtls/esp_crt_bundle/include -I/home/mac/esp/esp-idf-v4.3.3/components/app_update/include -I/home/mac/esp/esp-idf-v4.3.3/components/spi_flash/include -I/home/mac/esp/esp-idf-v4.3.3/components/bootloader_support/include -I/home/mac/esp/esp-idf-v4.3.3/components/nvs_flash/include -I/home/mac/esp/esp-idf-v4.3.3/components/pthread/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_gdbstub/include -I/home/mac/esp/esp-idf-v4.3.3/components/esp_gdbstub/xtensa -I/home/mac/esp/esp-idf-v4.3.3/components/esp_gdbstub/esp32 -I/home/mac/esp/esp-idf-v4.3.3/components/wpa_supplicant/include -I/home/mac/esp/esp-idf-v4.3.3/components/wpa_supplicant/port/include -I/home/mac/esp/esp-idf-v4.3.3/components/wpa_supplicant/include/esp_supplicant -I/home/mac/esp/esp-idf-v4.3.3/components/perfmon/include -I/home/mac/esp/esp-idf-v4.3.3/components/asio/asio/asio/include -I/home/mac/esp/esp-idf-v4.3.3/components/asio/port/include -I/home/mac/esp/esp-idf-v4.3.3/components/cbor/port/include -I/home/mac/esp/esp-idf-v4.3.3/components/unity/include -I/home/mac/esp/esp-idf-v4.3.3/components/unity/unity/src -I/home/mac/esp/esp-idf-v4.3.3/components/cmock/CMock/src -I/home/mac/esp/esp-idf-v4.3.3/components/coap/port/include -I/home/mac/esp/esp-idf-v4.3.3/components/coap/port/include/coap -I/home/mac/esp/esp-idf-v4.3.3/components/coap/libcoap/include -I/home/mac/esp/esp-idf-v4.3.3/components/coap/libcoap/include/coap2 -I/home/mac/esp/esp-idf-v4.3.3/components/console -I/home/mac/esp/esp-idf-v4.3.3/components/nghttp/port/include -I/home/mac/esp/esp-idf-v4.3.3/components/nghttp/nghttp2/lib/includes -I/home/mac/esp/esp-idf-v4.3.3/components/esp-tls -I/home/m [... cut ...]
/media/mac/sec-vol/workspace-esp/code_recognition/main/include/app_peripherals.h:9:10: fatal error: screen_driver.h: No such file or directory
 #include "screen_driver.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
[4/12] Performing build step for 'bootloader'
ninja: no work to do.
[5/10] Linking C static library esp-idf/libsodium/liblibsodium.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
mac@machine:/media/mac/sec-vol/workspace-esp/code_recognition$ 

```

Is there a standard IDF version which will work for all the examples in `esp-who`?

Auroragan, 2022-06-24:
please follow the steps in readme: https://github.com/espressif/esp-who#get-esp-idf

prabhu012, 2022-06-28:
when I build the code I am getting error

C:\esp\esp-who\examples\motion_detection\lcd>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp\esp-who\examples\motion_detection\lcd\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/esp/esp-who/examples/motion_detection/lcd/sdkconfig
Loading defaults file C:/esp/esp-who/examples/motion_detection/lcd/sdkconfig.defaults...
Loading defaults file C:/esp/esp-who/examples/motion_detection/lcd/sdkconfig.defaults.esp32s3...
C:/esp/esp-who/examples/motion_detection/lcd/sdkconfig.defaults.esp32s3:3 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
C:/esp/esp-who/examples/motion_detection/lcd/sdkconfig.defaults.esp32s3:4 CONFIG_ESP32S3_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
-- App "motion_detection_lcd" version: v0.9.3-167-g7563e00
-- Adding linker script C:/esp/esp-who/examples/motion_detection/lcd/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt bus cmock console cxx driver efuse esp-code-scanner esp-dl esp-sr esp-tls esp32-camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py fatfs fb_gfx freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mdns modules mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread screen sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/esp/esp-idf/components/app_trace C:/esp/esp-idf/components/app_update C:/esp/esp-idf/components/asio C:/esp/esp-idf/components/bootloader C:/esp/esp-idf/components/bootloader_support C:/esp/esp-idf/components/bt C:/esp/esp-who/components/bus C:/esp/esp-idf/components/cmock C:/esp/esp-idf/components/console C:/esp/esp-idf/components/cxx C:/esp/esp-idf/components/driver C:/esp/esp-idf/components/efuse C:/esp/esp-who/components/esp-code-scanner C:/esp/esp-who/components/esp-dl C:/esp/esp-who/components/esp-sr C:/esp/esp-idf/components/esp-tls C:/esp/esp-who/components/esp32-camera C:/esp/esp-idf/components/esp_adc_cal C:/esp/esp-idf/components/esp_common C:/esp/esp-idf/components/esp_eth C:/esp/esp-idf/components/esp_event C:/esp/esp-idf/components/esp_gdbstub C:/esp/esp-idf/components/esp_hid C:/esp/esp-idf/components/esp_http_client C:/esp/esp-idf/components/esp_http_server C:/esp/esp-idf/components/esp_https_ota C:/esp/esp-idf/components/esp_https_server C:/esp/esp-idf/components/esp_hw_support C:/esp/esp-idf/components/esp_lcd C:/esp/esp-idf/components/esp_local_ctrl C:/esp/esp-idf/components/esp_netif C:/esp/esp-idf/components/esp_phy C:/esp/esp-idf/components/esp_pm C:/esp/esp-idf/components/esp_psram C:/esp/esp-idf/components/esp_ringbuf C:/esp/esp-idf/components/esp_rom C:/esp/esp-idf/components/esp_serial_slave_link C:/esp/esp-idf/components/esp_system C:/esp/esp-idf/components/esp_timer C:/esp/esp-idf/components/esp_wifi C:/esp/esp-idf/components/espcoredump C:/esp/esp-idf/components/esptool_py C:/esp/esp-idf/components/fatfs C:/esp/esp-who/components/fb_gfx C:/esp/esp-idf/components/freertos C:/esp/esp-idf/components/hal C:/esp/esp-idf/components/heap C:/esp/esp-idf/components/http_parser C:/esp/esp-idf/components/idf_test C:/esp/esp-idf/components/ieee802154 C:/esp/esp-idf/components/json C:/esp/esp-idf/components/log C:/esp/esp-idf/components/lwip C:/esp/esp-who/examples/motion_detection/lcd/main C:/esp/esp-idf/components/mbedtls C:/esp/esp-idf/components/mdns C:/esp/esp-who/components/modules C:/esp/esp-idf/components/mqtt C:/esp/esp-idf/components/newlib C:/esp/esp-idf/components/nvs_flash C:/esp/esp-idf/components/openthread C:/esp/esp-idf/components/partition_table C:/esp/esp-idf/components/perfmon C:/esp/esp-idf/components/protobuf-c C:/esp/esp-idf/components/protocomm C:/esp/esp-idf/components/pthread C:/esp/esp-who/components/screen C:/esp/esp-idf/components/sdmmc C:/esp/esp-idf/components/soc C:/esp/esp-idf/components/spi_flash C:/esp/esp-idf/components/spiffs C:/esp/esp-idf/components/tcp_transport C:/esp/esp-idf/components/tinyusb C:/esp/esp-idf/components/ulp C:/esp/esp-idf/components/unity C:/esp/esp-idf/components/usb C:/esp/esp-idf/components/vfs C:/esp/esp-idf/components/wear_levelling C:/esp/esp-idf/components/wifi_provisioning C:/esp/esp-idf/components/wpa_supplicant C:/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
[... cut ...] .obj.d -o esp-idf/bus/CMakeFiles/__idf_bus.dir/spi_bus.c.obj -c C:/esp/esp-who/components/bus/spi_bus.c
C:/esp/esp-who/components/bus/spi_bus.c:19:10: fatal error: driver/spi_master.h: No such file or directory
   19 | #include "driver/spi_master.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[855/932] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_softap.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Rohit7077, 2024-07-19:
Hello,

I'm trying the AHT10 sensor data equitation in Espressif ide in multiple files like .h .c and main after compile i got thi error 
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c/i2c.c.obj 
ccache C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"GIT-NOTFOUND\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I"C:/Users/Rohit Salavi/workspace/AsPerStandard/build/config" -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/deprecated -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/analog_comparator/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/dac/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/gpio/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/gptimer/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2c/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2s/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/ledc/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/mcpwm/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/parlio/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/pcnt/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/rmt/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/sdio_slave/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/sdmmc/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/sigma_delta/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/spi/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/temperature_sensor/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/touch_sensor/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/twai/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/uart/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/usb_serial_jtag/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/touch_sensor/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/config/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/config/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/config/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/esp_additions/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/heap/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/log/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/soc/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/hal/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_system/port/include/private -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/xtensa/deprecated_include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/port/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/port/freertos/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/port/esp32xx/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/port/esp32xx/include/arch -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/lwip/port/esp32xx/include/sys -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v5.2.2/components/efuse/include -IC:/Espressif/fra [... cut ...] ch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c/i2c.c.obj -MF esp-idf\driver\CMakeFiles\__idf_driver.dir\i2c\i2c.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c/i2c.c.obj -c C:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2c/i2c.c
C:/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2c/i2c.c:25:10: fatal error: driver/i2c.h: No such file or directory
   25 | #include "driver/i2c.h"
      |          ^~~~~~~~~~~~~~

100312dog, 2024-07-25:
@Rohit7077 hi, you can try another version of idf.