espressif/esp-who #221: ESP-Eye Wakeup

mtc-20, 2022-06-29:
Hello,

I have some doubt regarding the ESP-EYE board. The product page shows that this board has a  mic for speech recognition. But I cannot find any of these features or voice recognition models in the `master` branch for ESP-EYE, only in ESP32-S3_EYE. Other websites that talk of ESP-EYE use an older branch of ESP_WHO which has examples of a voice wakeup and face recognition exmaple. I would like to understand why these examples are not available in the master branch anymore.

Is there any particular reason this was removed? I would like to buy the ESP-EYE for face recognition and speech recognition. But the fact that the latest documentation doesn't talk about any of this is a little concerning. I would like to be sure before making such an investment since this would be my first project with ESP32s.

liminghui-sandra, 2022-06-30:
Hi, we assume it's because of performance consideration. As Espressif has released ESP32-S3 and it SW support, we'd prefer to do AI applications on S3 related boards: S3-EYE, S3-Korvo-1, S3-Korvo-2, S3-BOX, etc. Maybe you can take a look: https://www.espressif.com/en/products/devkits

mtc-20, 2022-06-30:
Hi, thanks for that. But it's disappointing news for me. The reason why I wanted to buy the ESP-EYE was for the size, camera and mic. I can't find any exact replacement board in that same size. 
The S3-EYE, S3-Korvo-2 are more bulkier and comes with a LCD sub board assembled (and more expensive).

I think I will drop my current project idea for now.

Thanks for the info

closed 2022-09-21 by Auroragan