espressif/esp-who #227: build fail for ESP32-S3-EYE

sangbae, 2022-08-24:
Hello espressif
Recently, i bought ESP32-S3-EYE v2.2 and try to use esp-who as a base for my project. 
However, i found out that it's impossible to re-build default-firmware-binary from this repo. 
default binary flashed in ESP32-S3-EYE was built with esp-idf v5.0-dev and App version v0.9.4
**but i cannot find esp-who v0.9.4.** 

esp-who v0.9.3 with esp-idf v4.4.2 is not working as it is described in GettingStarted.

Could you please give me any help?

Thanks in advance.
-----------------------------------------------------------------------------------------------------------
[log messages]
1) pre-built default firmware's log message 
2) newly built firmware log message with esp-idf v5.0-dev and esp-who v0.9.3
-----------------------------------------------------------------------------------------------------------
1) pre-built default firmware's log message 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (582) esp_image: segment 1: paddr=00302aec vaddr=3fc9bfESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40001c3b
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x18bc
load:0x403b6000,len:0xb58
load:0x403ba000,len:0x2fdc
entry 0x403b61e4
**I (25) boot: ESP-IDF v5.0-dev-163-g55a4cfcfb8 2nd stage bootloader**
I (25) boot: compile time 16:17:59
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 factory          factory app      00 00 00010000 003e8000
I (78) boot:  1 model            Unknown data     01 82 003f8000 0036b000
I (86) boot:  2 nvs              WiFi data        01 02 00763000 00004000
I (93) boot:  3 fr               Unknown data     01 06 00767000 00020000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=2f2ac4h (3091140) map
I (582) esp_image: segment 1: paddr=00302aec vaddr=3fc9bf40 size=04140h ( 16704) load
I (585) esp_image: segment 2: paddr=00306c34 vaddr=40374000 size=093e4h ( 37860) load
I (595) esp_image: segment 3: paddr=00310020 vaddr=42000020 size=c5c9ch (810140) map
I (719) esp_image: segment 4: paddr=003d5cc4 vaddr=4037d3e4 size=0eb5ch ( 60252) load
I (731) esp_image: segment 5: paddr=003e4828 vaddr=50000000 size=00010h (    16) load
I (740) boot: Loaded app from partition at offset 0x10000
I (741) boot: Disabling RNG early entropy source...
W (741) bootloader_random: RNG for ESP32-S3 not currently supported
W (748) FLASH: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (758) opi psram: vendor id : 0x0d (AP)
I (762) opi psram: dev id    : 0x02 (generation 3)
I (768) opi psram: density   : 0x03 (64 Mbit)
I (773) opi psram: good-die  : 0x01 (Pass)
I (778) opi psram: Latency   : 0x01 (Fixed)
I (783) opi psram: VCC       : 0x01 (3V)
I (787) opi psram: SRF       : 0x01 (Fast Refresh)
I (793) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (798) opi psram: BurstLen  : 0x01 (32 Byte)
I (803) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (809) opi psram: DriveStrength: 0x00 (1/1)
W (814) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (844) cpu_start: Starting app cpu, entry point is 0x40375658
0x40375658: call_start_cpu1 at /home/sblee/esp/esp-idf/components/esp_system/port/cpu_start.c:195 (discriminator 3)
[... 176 lines cut ...]
I (5028) MN: SINGLE_RECOGNITION: V4.5 CN; core: 0; (Jan 17 2022 17:07:58)

I (5028) MN: ---------------------SPEECH COMMANDS---------------------
I (5028) MN: Command ID0, phrase 0: ting zhi gong zuo
I (5028) MN: Command ID1, phrase 1: jin xian shi
I (5028) MN: Command ID2, phrase 2: ren lian shi bie
I (5028) MN: Command ID3, phrase 3: yi dong zhen ce
I (5028) MN: Command ID4, phrase 4: tian jia ren lian
I (5028) MN: Command ID5, phrase 5: shan chu ren lian
I (5028) MN: Command ID6, phrase 6: shi bie yi xia
I (5028) MN: Command ID7, phrase 7: zhi leng mo shi
I (5028) MN: Command ID8, phrase 8: song feng mo shi
I (5028) MN: Command ID9, phrase 9: jie neng mo shi
I (5028) MN: Command ID10, phrase 10: chu shi mo shi
I (5028) MN: Command ID11, phrase 11: jian kang mo shi
I (5028) MN: Command ID12, phrase 12: shui mian mo shi
I (5028) MN: Command ID13, phrase 13: da kai lan ya
I (5028) MN: Command ID14, phrase 14: guan bi lan ya
I (5028) MN: Command ID15, phrase 15: kai shi bo fang
I (5028) MN: Command ID16, phrase 16: zan ting bo fang
I (5028) MN: Command ID17, phrase 17: ding shi yi xiao shi
I (5028) MN: Command ID18, phrase 18: da kai dian deng
I (5028) MN: Command ID19, phrase 19: guan bi dian deng
I (5038) App/Speech: Ready
W (5068) eFuse: calibration efuse version does not match, set default version: 0

liminghui-sandra, 2022-08-26:
Hello, actually not very clear about your 'build fail' error. 
Have you tried with the default bin v2.2 and example here?
https://github.com/espressif/esp-who/tree/master/default_bin/esp32-s3-eye
https://github.com/espressif/esp-who/tree/master/examples/esp32-s3-eye

sangbae, 2022-08-26:
> nd out that it's imp

Hello.
Default-binary's log, which was flashed from factory, showed that esp-who's version is 0.9.4
But, this repo does not has 0.9.4.  the lastest tag is v0.9.3

Can you commit 0.9.4 patch which is the same version to factory-version?

liminghui-sandra, 2022-08-26:
Hello, 0.9.4 was not released, and the tag is too old now. But we believe the default bin put in github can be successfully flashed, could you tell us your concern?

closed 2023-04-10 by sangbae