espressif/esp-who #234: Max number of face that can be store in esp32s3 eye

abhi-ratnman, 2022-11-24:
Hi i was wondering if any one has tested the limit for max number of faces we can store for facial recognition. I am planning to use the board in an industrial product.

abhi-ratnman, 2023-02-03:
Hi is this still active product or a dead product ?

Sandra-lol, 2023-02-03:
Hi,
ESP32-S3-EYE is still alive.
The default bin in the board can support at most 227 face IDs.

abhi-ratnman, 2023-02-03:
@Sandra-lol thanks for the response . is there any provision to increase that to 500 faces? I am thinking of selecting the module for a product hence was doing research if at all we can use this board. 

really appreciate your response

Auroragan, 2023-02-17:
Hi,
it is possible, will update the library soon.

abhi-ratnman, 2023-02-18:
Hi @Auroragan, 
thanks for the response. By when can we expect the update?
We are planning a product development which is to start in 2-3 weeks.

Auroragan, 2023-02-20:
Will update this week.

Auroragan, 2023-02-22:
Hi, please check latest master.

closed 2023-02-24 by Auroragan