espressif/esp-who #235: Init PSRAM manually trowing failed to allocate frame buffers.

nikhil-robinson, 2022-11-30:
Hello there is it possible  to initialize the psram manually and use it with who examples ?  i am trying to initialize psram after boot up but camera frame failed to allocate. I haveesp32s3 with 8 mb  OCTAL  psram.

liminghui-sandra, 2022-12-01:
Hi, it seems more like a psram initialization issue, better to create an issue in IDF repo and discuss there)

nikhil-robinson, 2022-12-01:
I was able to fix the above issue with enabling `esp_spiram_add_to_heapalloc();`   but now i am getting some other issues.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x38 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403758e6
0x403758e6: xt_highint4 at /home/nikhil/esp/esp-idf/components/esp_system/port/soc/esp32s3/highint_hdl.S:57

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1670
load:0x403c9700,len:0xbb0
load:0x403cc700,len:0x2f0c
entry 0x403c9950
I (24) boot: ESP-IDF v4.4.3-173-g1f63dc70b8 2nd stage bootloader
I (25) boot: compile time 15:38:51
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 8MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 factory          factory app      00 00 00010000 003c0000
I (66) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (73) boot:  2 fr               Unknown data     01 06 003e0000 00020000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=6b078h (438392) map
I (172) esp_image: segment 1: paddr=0007b0a0 vaddr=3fc96da0 size=0401ch ( 16412) load
I (176) esp_image: segment 2: paddr=0007f0c4 vaddr=40374000 size=00f54h (  3924) load
I (178) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=98a94h (625300) map
I (298) esp_image: segment 4: paddr=00118abc vaddr=40374f54 size=11e40h ( 73280) load
I (322) boot: Loaded app from partition at offset 0x10000
I (323) boot: Disabling RNG early entropy source...
I (334) cpu_start: Pro cpu up.
I (334) cpu_start: Starting app cpu, entry point is 0x40375674
0x40375674: call_start_cpu1 at /home/nikhil/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (348) cpu_start: Pro cpu start user code
I (348) cpu_start: cpu freq: 160000000
I (348) cpu_start: Application information:
I (351) cpu_start: Project name:     play_sdcard_music
I (357) cpu_start: App version:      f82540e-dirty
I (362) cpu_start: Compile time:     Dec  1 2022 12:03:03
I (368) cpu_start: ELF file SHA256:  46f00cf88739d58c...
I (374) cpu_start: ESP-IDF:          v4.4.3-173-g1f63dc70b8
I (380) heap_init: Initializing. RAM available for dynamic allocation:
I (388) heap_init: At 3FC9E070 len 0004B6A0 (301 KiB): D/IRAM
I (394) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (401) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (407) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (414) spi_flash: detected chip: generic
I (418) spi_flash: flash io: dio
I (428) sleep: Configure to isolate all GPIO pins in sleep state
I (429) sleep: Enable automatic switching of GPIO sleep configuration
I (436) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (446) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (456) opi psram: vendor id : 0x0d (AP)
I (466) opi psram: dev id    : 0x02 (generation 3)
I (466) opi psram: density   : 0x03 (64 Mbit)
I (476) opi psram: good-die  : 0x01 (Pass)
I (476) opi psram: Latency   : 0x01 (Fixed)
I (486) opi psram: VCC       : 0x01 (3V)
I (486) opi psram: SRF       : 0x01 (Fast Refresh)
I (496) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (496) opi psram: BurstLen  : 0x01 (32 Byte)
I (506) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
[... 139 lines cut ...]
I (677) PLAY_SDCARD_MUSIC: [3.2] Create i2s stream to write data to codec chip
W (677) I2S: APLL not supported on current chip, use I2S_CLK_D2CLK as default clock source
I (697) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (697) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (707) MY_BOARD_V1_0: I2S0, MCLK output by GPIO0
I (707) PLAY_SDCARD_MUSIC: [3.3] Create mp3 decoder
I (717) MP3_DECODER: MP3 init
I (717) PLAY_SDCARD_MUSIC: [3.4] Register all elements to audio pipeline
I (727) PLAY_SDCARD_MUSIC: [3.5] Link it together [sdcard]-->fatfs_stream-->music_decoder-->i2s_stream-->[codec_chip]
I (737) AUDIO_PIPELINE: link el->rb, el:0x3fceaad8, tag:file, rb:0x3fceb824
I (747) AUDIO_PIPELINE: link el->rb, el:0x3fceb494, tag:dec, rb:0x3fced96c
I (757) PLAY_SDCARD_MUSIC: [ 4 ] Set up  event listener
I (757) PLAY_SDCARD_MUSIC: [4.1] Listening event from all elements of pipeline
I (767) PLAY_SDCARD_MUSIC: [4.2] Listening event from peripherals
I (777) who_camera: Camera module is ESP-S3-EYE
I (777) s3 ll_cam: DMA Channel=4
I (777) cam_hal: cam init ok
I (787) sccb: pin_sda 4 pin_scl 5
I (797) camera: Detected camera at address=0x30
I (797) camera: Detected OV2640 camera
I (797) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (887) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 8
I (887) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 11520, lines_per_half_buffer: 24, dma_buffer_size: 23040
I (897) cam_hal: buffer_size: 23040, Y half_buffer_size: 11520, node_buffer_size: 3840, node_cnt: 6, total_cnt: 10
```

nikhil-robinson, 2022-12-01:
What i am trying to achieve is run ADF with esp-who framework with audio playback and stream.

closed 2022-12-15 by nikhil-robinson