espressif/esp-who #240: IDF 5.0 support.

nikhil-robinson, 2023-01-24:
Hey there,

will ESP-WHO be supported on idf 5.0 as i am planning to migrate to version 5.0 can some one shed some light on this.

Yuyc2099, 2023-02-13:
I want to know, too.

Auroragan, 2023-03-02:
please check branch idfv5.0

SoniaZotz, 2023-05-01:
is ESP-WHO be supported on idf 5.0??

Sandra-lol, 2023-05-04:
> is ESP-WHO be supported on idf 5.0??

Please check https://github.com/espressif/esp-who/tree/idfv5.0

anlpereira, 2023-07-03:
I'm using idf 5.1 and it is not working.

WayneKeenan, 2023-08-22:
I forgot to `git submodule update --recursive --init` after switching branch.
I now compiles, flashes and runs.

v-3n0m, 2023-09-04:
@Auroragan  recently espressif released opencv as an component for esp32... what will be the outcome for esp-who. seeing that esp-who s not been updated properly nor the esp-dl is available as an component.

Sandra-lol, 2023-09-04:
> espressif released opencv as an component for esp32 
@v-3n0m  Hi, I don't quite understand this sentence, could you help explain or show the link?

project-x51, 2023-10-18:
It might be worth mentioning that the IDF has more than one branch in the documentation. It took me a while to realize this was the case.

BlueSkyB, 2023-11-15:
If you are using ESP-IDF release/v5.0 branch, please switch the ESP-WHO branch to the idfv5.0. And remember to pull and update the submodules of ESP-WHO.

BlueSkyB, 2024-08-02:
Currently, the master branch already supports IDF v5.0 by default.

closed 2024-08-02 by BlueSkyB