espressif/esp-who #243: E (1037) MFN: Flash is empty

PilnyTomas, 2023-02-16:
Board: ESP-S3-EYE -MB_V2.2
Hi, I got the esp-who installed alongside of esp-idf on branch release/v4.4 (and other projects, hopefully not relevant)
```
...esp]$ tree -L 1
.
├── esp-idf
├── other-projects...
└── esp-who
```
I was able to build and flash, but the serial output stopped at the line with `E (1037) MFN: Flash is empty`

**Steps to reproduce:**

1. clone idf, checkout 4.4, install
2. clone who, cd to face-recognition-terminal
3. follow the steps in the documentation to setup
4. compile, flash, monitor

**Complete log:**
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048deb
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17c8
load:0x403c9700,len:0xe48
load:0x403cc700,len:0x2f74
entry 0x403c9978
I (24) boot: ESP-IDF v4.4.3-347-g9ee3c8337d 2nd stage bootloader
I (25) boot: compile time 19:49:55
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 factory          factory app      00 00 00010000 003c0000
I (71) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (78) boot:  2 fr               Unknown data     01 06 003e0000 00020000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=189be8h (1612776) map
I (343) esp_image: segment 1: paddr=00199c10 vaddr=3fc947f0 size=03ee4h ( 16100) load
I (346) esp_image: segment 2: paddr=0019dafc vaddr=40374000 size=0251ch (  9500) load
I (351) esp_image: segment 3: paddr=001a0020 vaddr=42000020 size=ae150h (713040) map
I (465) esp_image: segment 4: paddr=0024e178 vaddr=4037651c size=0e2cch ( 58060) load
I (483) boot: Loaded app from partition at offset 0x10000
I (484) boot: Disabling RNG early entropy source...
I (484) opi psram: vendor id : 0x0d (AP)
I (488) opi psram: dev id    : 0x02 (generation 3)
I (494) opi psram: density   : 0x03 (64 Mbit)
I (499) opi psram: good-die  : 0x01 (Pass)
I (504) opi psram: Latency   : 0x01 (Fixed)
I (509) opi psram: VCC       : 0x01 (3V)
I (513) opi psram: SRF       : 0x01 (Fast Refresh)
I (519) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (524) opi psram: BurstLen  : 0x01 (32 Byte)
I (529) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (535) opi psram: DriveStrength: 0x00 (1/1)
W (540) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (551) spiram: Found 64MBit SPI RAM device
I (555) spiram: SPI RAM mode: sram 80m
I (559) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (566) cpu_start: Pro cpu up.
I (570) cpu_start: Starting app cpu, entry point is 0x403754c4
0x403754c4: call_start_cpu1 at /home/pilnyt/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (865) spiram: SPI SRAM memory test OK
I (874) cpu_start: Pro cpu start user code
I (874) cpu_start: cpu freq: 240000000
I (874) cpu_start: Application information:
I (874) cpu_start: Project name:     human_face_recognition_terminal
I (875) cpu_start: App version:      v0.9.3-167-g7563e00
I (875) cpu_start: Compile time:     Feb 14 2023 19:49:50
I (875) cpu_start: ELF file SHA256:  6d042c182f6d9596...
I (876) cpu_start: ESP-IDF:          v4.4.3-347-g9ee3c8337d
I (876) heap_init: Initializing. RAM available for dynamic allocation:
I (876) heap_init: At 3FC9C9E8 len 0004CD28 (307 KiB): D/IRAM
I (876) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (877) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (877) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (878) spi_flash: detected chip: generic
I (878) spi_flash: flash io: qio
I (886) sleep: Configure to isolate all GPIO pins in sleep state
I (886) sleep: Enable automatic switching of GPIO sleep configuration
I (886) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (887) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (887) who_camera: Camera module is ESP-S3-EYE
I (887) s3 ll_cam: DMA Channel=4
I (887) cam_hal: cam init ok
I (887) sccb: pin_sda 4 pin_scl 5
I (897) camera: Detected camera at address=0x30
I (897) camera: Detected OV2640 camera
I (897) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (967) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 8
I (967) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 11520, lines_per_half_buffer: 24, dma_buffer_size: 23040
I (967) cam_hal: buffer_size: 23040, half_buffer_size: 11520, node_buffer_size: 3840, node_cnt: 6, total_cnt: 10
I (967) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
I (967) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
I (977) cam_hal: cam config ok
I (977) ov2640: Set PLL: clk_2x: 1, clk_div: 3, pclk_auto: 1, pclk_div: 8
E (1037) MFN: Flash is empty

```

Auroragan, 2023-02-17:
Hi,
'E (1037) MFN: Flash is empty' this is not a real error, please ignore it.

And there is no serial output before you interact.
please try using the botton like this: https://github.com/espressif/esp-who/tree/master/examples/human_face_recognition#use-the-example

PilnyTomas, 2023-02-24:
The readme says 

> The output results of this example can be shown on an LCD screen or in your terminal.

How am I supposed to interact when the terminal is stuck with an error? Please clarify in the documentation.

Auroragan, 2023-02-24:
The error message has been updated to info at latest master branch. please check. 
Will update the document as well.

closed 2023-02-24 by Auroragan