espressif/esp-who #246: no example of who is working

helmutAtGithub, 2023-02-19:
Here is the output:
-- Configuring incomplete, errors occurred!
See also "/home/lumi/esp/esp-who/examples/human_face_detection/terminal/build/CMakeFiles/CMakeOutput.log".
HINT: The component mdns could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.                                                                                              
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.                                                               
Refer to the migration guide for more details about moved components.                                                                                                            
Refer to the build-system guide for more details about how components are found and included in the build

See I was looking for the component:
find . -name "*mdns*"                           
./esp-idf/components/openthread/openthread/tests/scripts/thread-cert/border_router/test_mdns_restart.py
./esp-idf/components/lwip/lwip/doc/mdns.txt
./esp-idf/components/lwip/lwip/test/unit/mdns
./esp-idf/components/lwip/lwip/test/unit/mdns/test_mdns.c
./esp-idf/components/lwip/lwip/test/unit/mdns/test_mdns.h
./esp-idf/components/lwip/lwip/src/include/lwip/apps/mdns_priv.h
./esp-idf/components/lwip/lwip/src/include/lwip/apps/mdns.h
./esp-idf/components/lwip/lwip/src/include/lwip/apps/mdns_opts.h
./esp-idf/components/lwip/lwip/src/apps/mdns
./esp-idf/components/lwip/lwip/src/apps/mdns/mdns.c
./esp-idf/docs/zh_CN/api-reference/protocols/mdns.rst
./esp-idf/docs/en/api-reference/protocols/mdns.rst
Here it is ???!!!
./esp-who/components/modules/web/app_mdns.h
./esp-who/components/modules/web/app_mdns.c
Then I did this:
idf.py add-dependency mdns
Executing action: add-dependency
Created "/home/lumi/esp/esp-who/examples/human_face_detection/web/main/idf_component.yml"
Successfully added dependency "espressif/mdns*" to component "main"
Done
Then I dis this again: idf.py set-target esp32
-- Configuring incomplete, errors occurred!
See also "/home/lumi/esp/esp-who/examples/human_face_detection/web/build/CMakeFiles/CMakeOutput.log".
HINT: The component esp_adc_cal could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.                                                                                       
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.                                                               
Refer to the migration guide for more details about moved components.                                                                                                            
Refer to the build-system guide for more details about how components are found and included in the build.                                                                       
cmake failed with exit code 1, output of the command is in the /home/lumi/esp/esp-who/examples/human_face_detection/web/build/log/idf_py_stderr_output_726126 and /home/lumi/esp/esp-who/examples/human_face_detection/web/build/log/idf_py_stdout_output_726126
And again: idf.py add-dependency esp_adc_cal
CMake Error at /home/lumi/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  ERROR: Cannot find versions of "espressif/esp_adc_cal" with version
  satisfying "*" for the current target "esp32"
Call Stack (most recent call first):
  /home/lumi/esp/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:8 (project)
-- Configuring incomplete, errors occurred!
See also "/home/lumi/esp/esp-who/examples/human_face_detection/web/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/lumi/esp/esp-who/examples/human_face_detection/web/build/log/idf_py_stderr_output_730875 and /home/lumi/esp/esp-who/examples/human_face_detection/web/build/log/idf_py_stdout_output_730875
Whats going on here ?

helmutAtGithub, 2023-02-19:
Then I looked for the component:
https://components.espressif.com/components?q=esp_adc_cal
Answer:  Components not found
Really thats a mess ... sorry

helmutAtGithub, 2023-02-19:
find . -name "*esp*adc*cal*"
./esp-idf/components/esp_adc/deprecated/esp_adc_cal_internal_legacy.h
./esp-idf/components/esp_adc/deprecated/esp32c3/esp_adc_cal_legacy.c
./esp-idf/components/esp_adc/deprecated/include/esp_adc_cal_types_legacy.h
./esp-idf/components/esp_adc/deprecated/include/esp_adc_cal.h
./esp-idf/components/esp_adc/deprecated/esp32/esp_adc_cal_legacy.c
./esp-idf/components/esp_adc/deprecated/esp_adc_cal_common_legacy.c
./esp-idf/components/esp_adc/deprecated/esp32s2/esp_adc_cal_legacy.c
./esp-idf/components/esp_adc/deprecated/esp32s3/esp_adc_cal_legacy.c
I do not give up .... all is deprecated ????

helmutAtGithub, 2023-02-19:
Where do I get this component ??? I tried a lot of them. 

So today again. I needed 3 hours more to run a who example. Nothing is working except the poor hello_world ....

Developing with the esp32eye will be very expensive. Because no good documentation. I can not recommend !

helmutAtGithub, 2023-02-19:
Please use the latest ESP-IDF on the [release/v4.4](https://github.com/espressif/esp-idf/tree/release/v4.4) branch.
Is not true I just tried it.
Why is this repository still online ? To much wrong documentation for esp-who.
I got the cam running with the Arduino IDE. I fixed the bug with board.txt and givee 9600 Baud to the Serial. Then it was running.

lijunru-hub, 2023-02-21:
1. please use the latest ESP-IDF on the release/v4.4， this is important. Use `idf.py --version` 
2. please make sure to use `git submodule update --init --recursive` in esp-dl and esp-idf

helmutAtGithub, 2023-02-21:
I got everything from scratch with:
https://github.com/espressif/esp-idf.git
git clone https://github.com/espressif/esp-who.git and

git submodule update --init --recursive 
in esp-dl and esp-idf. I did it for esp-idf, esp-who and inside esp-dl folder (but this is really not a must !!!)

this is the branch I select:
git fetch --all
git branch -r
  origin/HEAD -> origin/master
  origin/audio/stack_on_psram_v3.3
  origin/ble_mesh_release/esp-ble-mesh-v0.6.1
  origin/customer/11kv_v4.0_xiaomi
  origin/customer/maintain_v4.0_xiaomi_tsf_issue
  origin/master
  origin/release/v2.0
  origin/release/v2.1
  origin/release/v3.0
  origin/release/v3.1
  origin/release/v3.2
  origin/release/v3.3
  origin/release/v4.0
  origin/release/v4.1
  origin/release/v4.2
  origin/release/v4.3
  origin/release/v4.4
  origin/release/v5.0
Then in esp-idf:
git checkout -b origin/release/v4.4
........
then again to be sure ....
git submodule update --recursive --init with no effect

Then:
git branch -v
  master              045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'
* origin/release/v4.4 045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'
in esp-idf
.  ./export.sh ....

After this I do:
idf.py --version
ESP-IDF v5.1-dev-3462-g045163a2ec => it does not show 4.4 !!!!! What is wrong ????

then I move to folder ...
~/…/esp-who/examples/human_face_detection/web]
└─$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/lumi/esp/esp-who/examples/human_face_detection/web/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/lumi/esp/esp-who/examples/human_face_detection/web/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/lumi/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/lumi/esp/esp-who/examples/human_face_detection/web"...
-- Found Git: /usr/bin/git (found version "2.39.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/lumi/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/lumi/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/lumi/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
[... 25 lines cut ...]
  master              045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'
* origin/release/v4.4 045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'
 
then:
idf.py --version       
ESP-IDF v5.1-dev-3462-g045163a2ec .... it does not show v4.4 can you tell me why ?

Is it possible that it is it is a merge accident ..... ?

Look exactly now:
git branch -v                          
  master              045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'
* origin/release/v4.4 045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'

I think this is the problem !!! Look at the hash values: 045163a2ec
045163a2ec Merge branch 'docs/update_performance_chapters_for_c6_h2' into 'master'
045163a2ec docs/update_performance_chapters_for_c6_h2' into 'master

then again:
idf.py --version       
ESP-IDF v5.1-dev-3462-g045163a2ec .... it does not show v4.4 can you tell me why ?

And please try before giving an answer !

Another question it possible to get the esp32 toolchain. I prefer to use my own written Makfiles. For more control !

lijunru-hub, 2023-02-22:
please don't use `git checkout -b `
use `git checkout  release/v4.4`
Thanks for u patience.

helmutAtGithub, 2023-02-22:
Okay great man !!!
Thank you very much. git checkout  release/v4.4  is enough. I think I have the used -b in a wrong way.
I destroyed the branch !!!! I produced a merge accident ... :-( !!!!

And sorry I was not patient enough. You have been very patient with me. Thank you very much.

This is what I got now.
idf.py --version
ESP-IDF v4.4.4-148-g4c2afac355 

And the build works fine !!!
By the way is there a possibility to work with gnu make ?

closed 2023-02-22 by helmutAtGithub