espressif/esp-who #249: esp-who/examples/human_face_detection/web 报错

weilian1977, 2023-03-14:
I (277) opi psram: vendor id : 0x0d (AP)
I (277) opi psram: dev id    : 0x02 (generation 3)
I (277) opi psram: density   : 0x03 (64 Mbit)
I (280) opi psram: good-die  : 0x01 (Pass)
I (285) opi psram: Latency   : 0x01 (Fixed)
I (290) opi psram: VCC       : 0x01 (3V)
I (295) opi psram: SRF       : 0x01 (Fast Refresh)
I (300) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (306) opi psram: BurstLen  : 0x01 (32 Byte)
I (311) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (317) opi psram: DriveStrength: 0x00 (1/1)
W (322) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (332) spiram: Found 64MBit SPI RAM device
I (336) spiram: SPI RAM mode: sram 80m
I (341) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (348) cpu_start: Pro cpu up.
I (352) cpu_start: Starting app cpu, entry point is 0x40375568
0x40375568: call_start_cpu1 at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (367) cpu_start: Pro cpu start user code
I (367) cpu_start: cpu freq: 240000000
I (367) cpu_start: Application information:
I (367) cpu_start: Project name:     human_face_detection_web
I (368) cpu_start: App version:      v0.9.3-167-g7563e00-dirty
I (368) cpu_start: Compile time:     Mar 14 2023 18:43:36
I (368) cpu_start: ELF file SHA256:  91f4aca3aef901b5...
I (368) cpu_start: ESP-IDF:          v4.4.3-dirty
I (369) heap_init: Initializing. RAM available for dynamic allocation:
I (369) heap_init: At 3FCA7758 len 00041FB8 (263 KiB): D/IRAM
I (369) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (369) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (370) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (370) spi_flash: detected chip: generic
I (371) spi_flash: flash io: qio
I (375) sleep: Configure to isolate all GPIO pins in sleep state
I (375) sleep: Enable automatic switching of GPIO sleep configuration
I (375) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (376) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (386) pp: pp rom version: e7ae62f
I (386) net80211: net80211 rom version: e7ae62f
I (396) wifi:wifi driver task: 3fced63c, prio:23, stack:6656, core=0
I (396) system_api: Base MAC address is not set
I (396) system_api: read default base MAC address from EFUSE
I (396) wifi:wifi firmware version: 8cb87ff
I (396) wifi:wifi certification version: v7.0
I (396) wifi:config NVS flash: enabled
I (396) wifi:config nano formating: disabled
I (396) wifi:Init data frame dynamic rx buffer num: 32
I (396) wifi:Init management frame dynamic rx buffer num: 32
I (396) wifi:Init management short buffer num: 32
I (396) wifi:Init static tx buffer num: 16
I (396) wifi:Init tx cache buffer num: 32
I (396) wifi:Init static tx FG buffer num: 2
I (396) wifi:Init static rx buffer size: 1600
I (396) wifi:Init static rx buffer num: 10
I (396) wifi:Init dynamic rx buffer num: 32
I (396) wifi_init: rx ba win: 6
I (396) wifi_init: tcpip mbox: 32
I (396) wifi_init: udp mbox: 6
I (396) wifi_init: tcp mbox: 6
I (396) wifi_init: tcp tx win: 5744
I (396) wifi_init: tcp rx win: 5744
I (396) wifi_init: tcp mss: 1440
I (396) wifi_init: WiFi IRAM OP enabled
I (396) wifi_init: WiFi RX IRAM OP enabled
I (406) camera wifi: wifi_init_softap finished.SSID:Human Face Detection password:
I (406) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (436) wifi:mode : softAP (7c:df:a1:e1:a1:b9)
I (436) wifi:Total power save buffer number: 8
I (436) wifi:Init max length of beacon: 752/752
I (436) wifi:Init max length of beacon: 752/752
I (436) wifi:Set ps type: 0

I (436) who_camera: Camera module is ESP-S3-EYE
I (436) s3 ll_cam: DMA Channel=4
I (436) cam_hal: cam init ok
I (446) sccb: pin_sda 4 pin_scl 5
I (456) camera: Detected camera at address=0x3c
I (456) camera: Detected OV3660 camera
I (456) camera: Camera PID=0x3660 VER=0x00 MIDL=0x00 MIDH=0x00
I (766) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (766) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (766) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (766) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (766) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (766) cam_hal: cam config ok
I (776) ov3660: Calculated VCO: 120000000 Hz, PLLCLK: 60000000 Hz, SYSCLK: 15000000 Hz, PCLK: 7500000 Hz
I (796) camera_httpd: Starting web server on port: '80'
I (796) camera_httpd: Starting stream server on port: '81'
W (5286) cam_hal: Failed to get the frame on time!

***ERROR*** A stack overflow in task who_camera has been detected.

Backtrace: 0x40375e9a:0x3fcd15a0 0x4037de01:0x3fcd15c0 0x4038174e:0x3fcd15e0 0x4037fb1f:0x3fcd1660 0x4037deb8:0x3fcd1680 0x4037deae:0x00000000 |<-CORRUPTED
0x40375e9a: panic_abort at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/esp_system/panic.c:402

0x4037de01: esp_system_abort at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/esp_system/esp_system.c:128

0x4038174e: vApplicationStackOverflowHook at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/freertos/port/xtensa/port.c:394

0x4037fb1f: vTaskSwitchContext at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/freertos/tasks.c:3505

0x4037deb8: _frxt_dispatch at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/freertos/port/xtensa/portasm.S:436

0x4037deae: _frxt_int_exit at /home/weilian/esp-idf-v4.4.3_release_path_120mhz/components/freertos/port/xtensa/portasm.S:231

Raven888888, 2023-03-15:
I had the same problem with the default ESP-S3-EYE dev board and default camera

weilian1977, 2023-03-16:
ESP-S3-EYE dev board V1.1 only support 80MHz flash and spram？

weilian1977, 2023-03-19:
ESP-IDF: v4.4.3不会报这个错

weilian1977, 2023-03-19:
> ESP-IDF: v4.4.？

weilian1977, 2023-03-19:
> I had the same problem with the default ESP-S3-EYE dev board and default camera

ESP-IDF: v4.4.？

Raven888888, 2023-03-20:
I fixed it by this
```
diff --git a/driver/esp_camera.c b/driver/esp_camera.c
index dcb26ae..5994c52 100644
--- a/driver/esp_camera.c
+++ b/driver/esp_camera.c
@@ -305,7 +305,8 @@ esp_err_t esp_camera_deinit()
     return ret;
 }
 
-#define FB_GET_TIMEOUT (4000 / portTICK_PERIOD_MS)
+// #define FB_GET_TIMEOUT (4000 / portTICK_PERIOD_MS)
+#define FB_GET_TIMEOUT portMAX_DELAY
 
 camera_fb_t *esp_camera_fb_get()
 {

```

nikolausspence, 2023-03-26:
See 91b61dc other commit to fix stack overflow.  

In my case, I had to increase further to 4 * 1024 to eliminate the who_camera task stack overflow.   Once you set the camera clock settings through the web interface, the "cam_hal: Failed to get the frame on time!" error stops.   Would be nice to be more clear about how to set the default clock settings in code instead of through the web interface but the interface is very useful for finding the settings that finally produce a picture.

If changing the timeout alone does not stop the stack overflow, increase the stack size for the who_camera task in who_camera.c

mccoyg, 2023-04-15:
The above edits work. It was a little confusing the example in git code. So I will offer a clearer explanation.

in path/file: driver/esp_camera.c

find line: #define FB_GET_TIMEOUT (4000 / portTICK_PERIOD_MS)

comment out above line and/or replace with: #define FB_GET_TIMEOUT portMAX_DELAY