espressif/esp-who #251: Support need to Up the LCD!

Babu8008, 2023-04-05:
Hi team,

I am new to this ESP32-S3-EYE , I need to do programming on LCD in Arduino ide but i am unable to programming on it. i tried with example codes i.e TFT and TFT_espi examples also. 
I need part number of that 1.3 inch LCD and what are the drivers support for that LCD and pixels of that LCD kind of information.

please help me out.

Sandra-lol, 2023-04-06:
Hello, the LCD model is WT013-H12102 from wireless-tag
driver is here: https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/src/esp_lcd_panel_st7789.c

Babu8008, 2023-04-06:
Hi sandra-lol,
 
Thank you for your replay.
with this driver is it possible to do LCD programming in Arduino ide 2.0.4v platform?

Sandra-lol, 2023-04-06:
Yeah should work!

Babu8008, 2024-01-03:
Thanks for support

closed 2024-01-03 by Babu8008